nxp-mcuxpresso/legacy-mcux-sdk #7: Missing fsl_lpi2c_freertos driver files for PR.

Hadatko, 2021-03-10:
**Is your enhancement proposal related to a problem? Please describe.**
Hi, i have issue with fsl_lpi2c_freertos driver. We are using it and it is using dynamicly allocated freertos semaphore/mutex. I made a small changes to support also staticly allocated space. But i cannot find header and source file here. How can i push PR for this files?

**Describe the solution you'd like**
*header file:*
[image]

*source file:*
[image]

**Describe alternatives you've considered**
-

**Additional context**
-

nxp-susan, 2021-03-11:
Hi @Hadatko, appreciate for your contribution. However, peripheral drivers with middleware support are not included in this repository because currently the project does not include middleware. I have forward your suggestion to develop team for them to check, thanks.

Hadatko, 2021-03-11:
Hi @nxp-susan  thank you very much for forwarding code to development team. I just found out bug in my propose. I replaced mutex type by mistake. Take that into account please. (Non recursive with recursive)

Hadatko, 2021-03-11:
[image]

nxp-susan, 2021-03-11:
Hi @Hadatko, thanks for the update. I shared the link to develop team and they should be able to check your update.

Hadatko, 2021-03-11:
Thank you for quick response. I am looking forward for your outputs.

Hadatko, 2021-03-28:
Any update here?

DavidJurajdaNXP, 2021-05-20:
Hi @Hadatko , feel free to raise pull request. FreeRTOS support is available on github now.
https://github.com/NXPmicro/mcux-sdk/tree/main/drivers/lpi2c

Hadatko, 2021-05-20:
PR created.

closed 2024-02-19 by DavidJurajdaNXP

Hadatko, 2024-02-19:
Thank you for the info. Not happy, but understand.