nxp-mcuxpresso/legacy-mcux-sdk #8: Lpm.c is using only rtos functions with dynamicly allocated space.

Hadatko, 2021-03-11:
**Is your enhancement proposal related to a problem? Please describe.**
Hi, as i am reworking application to use static api only i have to modified lpm.c file 

**Describe the solution you'd like**
Ability to use static allocation api if chosen
[image]
[image]

**Describe alternatives you've considered**
-

**Additional context**
Similar to #7

nxp-susan, 2021-03-12:
Hi @Hadatko, thank you for the advice. The lpm.c file you mentioned in this issue is not included in current MCUXpresso SDK project. I will forward to develop team to check, appreciate for your patience.

Hadatko, 2021-03-12:
Hi @nxp-susan , no problem. I am preparing one more issue ticket related to srtm. But there is more complicated situation to implement static api.

Hadatko, 2021-03-28:
Any update here?

Hadatko, 2021-06-18:
@mcuxsusan i see that this lpm.c file is also located in https://github.com/NXPmicro/mcux-sdk-examples so i am moving there.

Hadatko, 2021-06-18:
Clossing as PR was created here: https://github.com/NXPmicro/mcux-sdk-examples/pull/4

closed 2021-06-18 by Hadatko