nxp-mcuxpresso/legacy-mcux-sdk #10: SRTM Freertos implementation doesn't support static api

Hadatko, 2021-03-12:
**Is your enhancement proposal related to a problem? Please describe.**
Only Freertos dynamic API is used. I would prefer completely static api usage.

**Describe the solution you'd like**
I will put several code changes here represent my current solution which you can consider.
*user srtm_config.h*
[image]
*srtm_defs.h*
[image]
*srtm_mutex_freertos.c*
[image]
*srtm_sem_freertos.c*
[image]
*srtm_lfcl_service.c*
[image]
[image]
*srtm_dispatcher_struct.h*
[image]
*srtm_message_struct.h*
[image]
*srtm_peercore_struct*
[image]
*srtm_peercore.c*
[image]
*srtm_mutex.h*
[image]
*srtm_sem.h*
[image]
*srtm_dispatcher.c*
[image]
[image]
[image]

**Describe alternatives you've considered**
-

**Additional context**
Similar to #7, #8, #9

Things need to be solved which are not yet covered: SRTM_Heap_Malloc and SRTM_Heap_Free

mcuxsusan, 2021-03-25:
Hi @Hadatko, would you like to directly improve our code by direct pull request? I have added more components on latest main branch, and the srtm component is now included.

Hadatko, 2021-03-25:
Sure. Thank you. Will do tomorrow.

Hadatko, 2021-03-28:
I am closing this issue as PR was created.

Hadatko, 2024-02-01:
Hi @mcuxsusan , i am reopening this issue as i noticed that not all requests were matched (i missed that last time, but i am doing update now and i see missing one file changes).
It looks like srtm_lfcl_service.c changes were not applied.
*srtm_lfcl_service.c*
[image]
[image]

But to match srtm static dynamic definition it should looks like:
[image]
[image]

mcuxsusan, 2024-02-02:
Hi @Hadatko, thanks for your active communication! I saw you have opened the pr, will inform the dev team to help review it. Hope we could get the pr merged soon.

Hadatko, 2024-02-02:
Hi @mcuxsusan thank you very much. Just note that the comment in this issue cannot be done by PR as file is not present in repository officialy. The PR you are mentioning is related to other srtm files which are present in this repository.

MichalPrincNXP, 2024-02-09:
Thank you Dusan, I do not know why this srtm_lfcl_service.c is not part of github release, but anyway, internal update requested and the fix will be provided in the next release. Can we close this ticket now?

MichalPrincNXP, 2024-02-09:
OK, let's wait once the internal PR is approved and merged internally and I will close the issue then.

Hadatko, 2024-02-09:
Sounds great 🤞

MichalPrincNXP, 2024-03-18:
Internal PR is approved and the fix is merged to be ready in the next MCUXpresso SDK 2.16.000 release

closed 2024-03-18 by MichalPrincNXP