nxp-mcuxpresso/legacy-mcux-sdk #12: Where can I find usb_device_dci.c?

stapelberg, 2021-03-24:
Thanks for providing the `mcux-sdk` via GitHub. It’s a very convenient and professional way of distributing your code! :)

I’m trying to use the SDK in my project, but I’m having some trouble with the USB part: files such as `usb_device_dci.c` which provide `USB_DeviceInit()` do not seem to be included in the `mcux-sdk` repository?

Are they in some other repository that I can’t find?

If they are not yet in any git repository, do you have plans to publish them, too?

Thanks,

mcuxsusan, 2021-03-25:
Yes, middleware usb stack code currently is included in the project delivery. Currently for MCUXpresso SDK delivery, we only have two project: mcux-sdk and mcux-sdk-examples.  USB is part of the mcux-middleware projects and we plan to add it later, but I have no ETA when we will launch that on GitHub, appreciate for your understanding.

stapelberg, 2021-03-25:
Thank you for letting me know.

It’s great to hear that you are planning to put `mcux-middleware` on GitHub eventually! Looking forward to it.

closed 2021-03-25 by mcuxsusan