nxp-mcuxpresso/legacy-mcux-sdk #17: Add into cmakelist option to print memory usage in %

Hadatko, 2021-04-14:
**Is your enhancement proposal related to a problem? Please describe.**
I think users generally appreciate this information.

**Describe the solution you'd like**
Add following lines to the CmakeList by default.
SET(CMAKE_EXE_LINKER_FLAGS_DEBUG "${CMAKE_EXE_LINKER_FLAGS_DEBUG} -Wl,--print-memory-usage")
SET(CMAKE_EXE_LINKER_FLAGS_RELEASE "${CMAKE_EXE_LINKER_FLAGS_RELEASE} -Wl,--print-memory-usage")
[image]

**Describe alternatives you've considered**
-

**Additional context**
Ussually i think i saw this output in other IDEs (i may be wrong). I think i saw it also in MCUXpresso IDE (not sure if i had to check some option there).

mcuxsusan, 2021-04-19:
The improvement sounds good, checking with the team about your proposal. Appreciate for your always input!

mcuxsusan, 2021-04-21:
Status update: the improvement has been done and will deploy in next MCUX SDK mainline release, which plans in mid July.

Hadatko, 2021-04-21:
Thank you very much ;)

closed 2021-04-21 by Hadatko