nxp-mcuxpresso/legacy-mcux-sdk #23: Fixing  -Wo-address-of-packed-member warnings (armgcc >= 9)

Hadatko, 2021-06-18:
**Is your enhancement proposal related to a problem? Please describe.**
From armgcc version 9 code belllow is fixing -Wo-address-of-packed-member warnings. Files are not present in repository.

**Describe the solution you'd like**
Use temporary variable for getting output data.
`srtm_pmic_service.c`
[image]
[image]

`srtm_rtc_service.c`
[image]
[image]

**Describe alternatives you've considered**

**Additional context**
[image]

mcuxsusan, 2021-06-18:
@Hadatko, thanks for raising the issue. This file is distributed in mcux-sdk-examples project and not inside this core project. For example, you may find the srtm_pmic_service.c at link https://github.com/NXPmicro/mcux-sdk-examples/blob/main/evkmcimx7ulp/demo_apps/power_mode_switch/srtm/services/srtm_pmic_service.c.

I will ask developer to check if your fix is OK. Appreciate for your contribution.

Hadatko, 2021-06-18:
@mcuxsusan Thank you for quick response. I forgot the other repository. I will create PR there and i am closing this. I was confused by: https://github.com/NXPmicro/mcux-sdk/tree/main/components/srtm/services

Hadatko, 2021-06-18:
Closing as PR was created: 
https://github.com/NXPmicro/mcux-sdk-examples/pull/3

closed 2021-06-18 by Hadatko