nxp-mcuxpresso/legacy-mcux-sdk #28: [BUG] Application freeze in DbgConsole_Flush when many bigger messages are sent

Hadatko, 2021-07-16:
**Describe the bug**
Here is modified example for imx7ulp [erpc_matrix_multiply_rpmsg_rtos_imxcm4.zip](https://github.com/NXPmicro/mcux-sdk/files/6831754/erpc_matrix_multiply_rpmsg_rtos_imxcm4.zip).  Changes are in `source/main_core1.c` and cmake related files (flags for release target 
    -DDEBUG_CONSOLE_TRANSFER_NON_BLOCKING \
    -DPRINTF_ADVANCED_ENABLE=1 \
    -D__FILENAME__='\"$(notdir $(abspath $<))\"' \)
 Zip file contains also `release` dir with built binaries. If few printf are removed from modification, program is working. But with all modification program will freeze.

**To Reproduce**
[image]
described above

**Expected behavior**
To not freeze at printf function.

**Screenshots and console output**
[image]

Hadatko, 2021-07-16:
Without this modification `-DDEBUG_CONSOLE_TRANSFER_NON_BLOCKING \` demo works. So combination usage of this macro and a many big prints usage result in app freeze.

Hadatko, 2021-07-16:
I think putting DbgConsole_Flush after printf when this macro is used made it freezing too.

mcuxsusan, 2021-07-19:
Hi @Hadatko, thank you for reporting the issue. To summarize the situation, do you mean that:
1. The demo in package works fine. 
2. With modification of flags.cmake for release target, adding macro definition 
-DDEBUG_CONSOLE_TRANSFER_NON_BLOCKING
-DPRINTF_ADVANCED_ENABLE=1
-D__FILENAME__='"$(notdir $(abspath $<))"' ),  the application run to freeze.
3.  Without modification of -DDEBUG_CONSOLE_TRANSFER_NON_BLOCKING in flags.cmake, the demo works.

So, the key point may be: using -DDEBUG_CONSOLE_TRANSFER_NON_BLOCKING in flags.cmake makes the demo run to freeze. And when uses -DDEBUG_CONSOLE_TRANSFER_NON_BLOCKING,  reduce use of printf or adding DbgConsole_Flush after printf both could improve the situation and demo works. Please help clarify if my understanding is correct. Appreciate for your energetic contribution. For your information, you could also follow the guide in https://github.com/NXPmicro/mcux-sdk/#overview to obtain the whole GitHub SDK distribution via zephyr west tool, then you could be able to also get the examples delivery in https://github.com/NXPmicro/mcux-sdk-examples/ and no need to MCUXpresso SDK package to reproduce your issue.

Hadatko, 2021-07-19:
Hi @mcuxsusan , The main issue is combination of -DDEBUG_CONSOLE_TRANSFER_NON_BLOCKING and many printf function in main_core1.c. Using DbgConsole_Flush freezing too may be additional or same issue. Not sure, but in current state not important.

mcuxsusan, 2021-07-19:
@lylezhu2014, may you please help comment on the situation?

Hadatko, 2021-12-15:
Any news here?

mcuxsusan, 2021-12-15:
Hi @Hadatko, thanks for your patience. @lylezhu2014, please help update your progress here.

lylezhu2014, 2021-12-17:
Hi  Hadatko,

The PRINTF and DbgConsole_Flush should be called in task context. Because FreeRTOS resource is used by debug console.

Hadatko, 2022-01-04:
Thank you very much. I moved more context into tasks and rest i removed. This is working well.

closed 2022-01-04 by Hadatko