nxp-mcuxpresso/legacy-mcux-sdk #29: [BUG] Using the MCUX SDK as a library seems to not add boot header

danewalton, 2021-07-20:
**Describe the bug**
I had added a PR a while back which makes the include headers public so that the mcux sdk components could be used as a library instead of "decorating" an executable. This allows for multiple binaries to have the library linked in the same CMake file. When we do that, it seems that the boot header isn't properly inserted and the image won't flash.

**To Reproduce**
* Environment (please complete the following information):
   - Tag/Commit hash: latest
   - Toolchain: gcc-arm-none-eabi-10-2020-q4-major
   - Board/SoC: mimxrt1060
   - Example: own
* Steps to reproduce the behavior:

Example CMake here with the issue. You can maybe try doing what is here with samples in your sample repo and just build the mcux as a library. Then link it to two samples.

```
# Copyright (c) Microsoft Corporation. All rights reserved.
# SPDX-License-Identifier: MIT

# set parent scope path
set(BOARD_DEMO_CONFIG_PATH ${CMAKE_CURRENT_SOURCE_DIR}/config CACHE INTERNAL "Config path")
set(BOARD_DEMO_FREERTOS_PORT_PATH ${FreeRTOS_ARM_CM4F_PATH} CACHE INTERNAL "FreeRTOS Port used ")

# Define
add_definitions(-DFSL_FEATURE_PHYKSZ8081_USE_RMII50M_MODE=1 -DSDK_DEBUGCONSOLE=1 -DSDK_DEBUGCONSOLE_UART -DLWIP_DHCP=1 -DLWIP_DNS=1 -DUSE_RTOS=1 -DFSL_RTOS_FREE_RTOS -DLWIP_TIMEVAL_PRIVATE=0 -D__STARTUP_CLEAR_BSS -D__STARTUP_INITIALIZE_NONCACHEDATA)

include(${CMAKE_CURRENT_SOURCE_DIR}/gcc_flags.cmake)

set(MCUX_SDK_PROJECT_NAME mcux-sdk-lib)

add_library(${MCUX_SDK_PROJECT_NAME})
target_include_directories(${MCUX_SDK_PROJECT_NAME}
  PUBLIC
    ${BOARD_DEMO_FREERTOS_PORT_PATH}
)
target_link_libraries(${MCUX_SDK_PROJECT_NAME}
  PUBLIC
    FreeRTOS
)

nxp_mcux_sdk_fetch()
lwip_fetch()

list(APPEND CMAKE_MODULE_PATH ${NXP_MCUX_SDK_PATH})

find_package(LWIP)

include_directories(${BOARD_DEMO_CONFIG_PATH})
include_directories(nxp_code)
include_directories(nxp_code/lwip)

file(GLOB NXPCODE_SOURCES nxp_code/*.c nxp_code/lwip/*.c)
set(PROJECT_SOURCES ${NXPCODE_SOURCES} main.c)

# configure modules
set(CONFIG_USE_driver_lpuart true)
set(MCUX_DEVICE "MIMXRT1062")

# Include sdk
include(all_devices)

include(driver_mdio-enet)

include(driver_phy-device-ksz8081)

# include modules
include(driver_igpio)

include(driver_clock)

include(driver_common)

include(device_CMSIS)

include(utility_debug_console)

include(component_lpuart_adapter)

include(component_serial_manager)

include(component_lists)

include(component_serial_manager_uart)

include(driver_lpuart)

include(device_startup)

include(driver_iomuxc)

include(utility_assert)

include(driver_xip_device)

include(driver_xip_board)

include(CMSIS_Include_core_cm7)

include(driver_mdio-common)

include(CMSIS_Include_common)

include(CMSIS_Include_dsp)

include(utilities_misc_utilities)

include(device_system)

include(driver_enet)

include(driver_trng)

include(driver_phy-common)

include(driver_dcp)

add_executable(${PROJECT_NAME} ${PROJECT_SOURCES})
target_link_libraries(${PROJECT_NAME} PRIVATE
    FreeRTOS::Timers
    FreeRTOS::Heap::5
    FreeRTOS::ARM_CM4F
    FreeRTOS::EventGroups
    FreeRTOSPlus::Utilities::backoff_algorithm
    FreeRTOSPlus::Utilities::logging
    FreeRTOSPlus::ThirdParty::mbedtls
    az::iot_middleware::freertos
    LWIP
    SAMPLE::SOCKET::LWIP
    SAMPLE::AZUREIOT
    SAMPLE::TRANSPORT::MBEDTLS
    ${MCUX_SDK_PROJECT_NAME}
    )

add_custom_command(TARGET ${PROJECT_NAME}
    # Run after all other rules within the target have been executed
    POST_BUILD
    COMMAND ${CMAKE_OBJCOPY} -Obinary $<TARGET_FILE:${PROJECT_NAME}> ${PROJECT_NAME}.bin
    COMMENT "Generate Bin file"
    VERBATIM)

add_executable(${PROJECT_NAME}-pnp ${PROJECT_SOURCES})
target_link_libraries(${PROJECT_NAME}-pnp PRIVATE
    FreeRTOS::Timers
    FreeRTOS::Heap::5
    FreeRTOS::ARM_CM4F
    FreeRTOS::EventGroups
    FreeRTOSPlus::Utilities::backoff_algorithm
    FreeRTOSPlus::Utilities::logging
    FreeRTOSPlus::ThirdParty::mbedtls
    az::iot_middleware::freertos
    LWIP
    SAMPLE::SOCKET::LWIP
    SAMPLE::AZUREIOTPNP
    SAMPLE::TRANSPORT::MBEDTLS
    ${MCUX_SDK_PROJECT_NAME}
    )

add_custom_command(TARGET ${PROJECT_NAME}-pnp
    # Run after all other rules within the target have been executed
    POST_BUILD
    COMMAND ${CMAKE_OBJCOPY} -Obinary $<TARGET_FILE:${PROJECT_NAME}-pnp> ${PROJECT_NAME}-pnp.bin
    COMMENT "Generate Bin file"
    VERBATIM)
```

**Expected behavior**
I would expect that the map files would be equivalent for the boot section.

**Screenshots and console output**
Working:
[image]

Not working:
[image]

**Additional context**
Add any other context about the problem here.

mcuxsusan, 2021-07-21:
Hi @danewalton, thanks for sharing your experience compiling include components into mcux-sdk-lib and linking the lib to your executables. However, it seems that the boot header not exist issue in your map file is not a BUG. For flash target, you need to add the macro definition 
    -DXIP_EXTERNAL_FLASH=1 \
    -DXIP_BOOT_HEADER_ENABLE=1 \
in CMAKE_C_FLAGS, this two macros controls the source of driver_xip_board and driver_xip_device for boot_hdr related definitions.
You could reference https://github.com/NXPmicro/mcux-sdk-examples/blob/main/evkmimxrt1060/demo_apps/hello_world/armgcc/flags.cmake for an example of flags setting for different targets.

hihigupt, 2021-07-21:
Hi
@mcuxsusan 

Thanks for looking into it.

Our samples that uses mcux-sdk do passes these macro when compiling (https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/26e7828674de19fb2c2941a3e9e84fb863ccbe50/demos/projects/NXP/mimxrt1060/gcc_flags.cmake#L17).

Issue here is linker is dropping these boot_hdr from the final binary. Digging deep into this, linker is dropping these un-used section when object files that contains these sections are part of static library. Same thing works fine if these object files are feed to linker directly, with KEEP syntax in linker script file.  Following stackoverflow explains the problem in detail:

https://stackoverflow.com/questions/44675582/gcc-how-to-tell-linker-not-to-skip-unused-sections

Right now to unblock, I am building mcux-sdk library, and passing all the files required for XIP as part of final linking steps (solution 3 from the stack overflow). It would be great if you guys can provide some support for this problem. Possible solution can be:

1. Add in linker script these boot header variable, so linker knows to link them.
2. Possibly provide list of all files required for boot header, via some variables. So people can link to these file as part of binary creation.
3.  Explicitly stating this problem in your readme and tell what files to include when.

Thanks,
Himanshu

mcuxsusan, 2021-07-22:
Hi @hihigupt, thanks for clarification. I was not aware that the reported issue caused by missing KEEP section descriptions in linker file because for XIP targets our linker file already KEEP these sections. See our example linker file for flash targets: https://github.com/NXPmicro/mcux-sdk-examples/blob/main/evkmimxrt1060/demo_apps/hello_world/armgcc/MIMXRT1062xxxxx_flexspi_nor.ld. 
So,  currently we have already used your suggested solution 1 in XIP linker files to keep these sections.  But for RAM targets, these sections are not needed because our RAM target directly link and download to RAM region, so these flash boot headers are not needed, then KEEP is not used to keep these sections. Please let me know if current way describing this in XIP linker file is enough, appreciate for your further input to suggest how should we do this better.

hihigupt, 2021-07-22:
Hi @mcuxsusan 

In both the cases KEEP is there in the ld script for boot_hdr (https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/26e7828674de19fb2c2941a3e9e84fb863ccbe50/demos/projects/NXP/mimxrt1060/MIMXRT1062xxxxx_sdram.ld#L61). But in static library case where boot_hdr section is inside object file that is part of static library, linker does not bring this section to final binary. Reason for not bringing this section to final binary is explained here: https://stackoverflow.com/questions/44675582/gcc-how-to-tell-linker-not-to-skip-unused-sections

'
As you know, a static library is an ar archive of object files.

Suppose libfoobar.a contains just foo.o and bar.o. A linkage:

g++ -o prog a.o foo.o bar.o     # A
is not the same as the linkage:

g++ -o prog a.o -lfoobar.   # B
The linker unconditionally consumes every object file in the linkage sequence, so in case A, it links a.o, foo.o, bar.o in prog.

The linker does not unconditionally consume every object file that is a member of a static library in the linkage sequence. A static library is a way of offering to the linker a bunch of object files from which to pick the ones it needs.
'

hihigupt, 2021-07-22:
Hi @mcuxsusan 

In both the cases KEEP is there in the ld script for boot_hdr (https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/26e7828674de19fb2c2941a3e9e84fb863ccbe50/demos/projects/NXP/mimxrt1060/MIMXRT1062xxxxx_sdram.ld#L61). But in case where boot_hdr section is inside static library, linker does not bring this section to final binary. Reason for not bringing this section to final binary is explained here: https://stackoverflow.com/questions/44675582/gcc-how-to-tell-linker-not-to-skip-unused-sections

'
As you know, a static library is an ar archive of object files.

Suppose libfoobar.a contains just foo.o and bar.o. A linkage:

g++ -o prog a.o foo.o bar.o     # A
is not the same as the linkage:

g++ -o prog a.o -lfoobar.   # B
The linker unconditionally consumes every object file in the linkage sequence, so in case A, it links a.o, foo.o, bar.o in prog.

The linker does not unconditionally consume every object file that is a member of a static library in the linkage sequence. A static library is a way of offering to the linker a bunch of object files from which to pick the ones it needs.
'

mcuxsusan, 2021-07-26:
Hi @hihigupt, thanks for your correction.  I am sorry I had some misunderstanding of the issue and the feedback is delayed. 
I have reproduced the boot header section missing issue when linking application to the built component library, it's a BUG and I have not carefully tested the situation. The answer in your shared link is very helpful, it describes why the issue happens and gives useful suggestions for resolution. I have tried all 3 solutions and they all works well.
1. Adding below linker options in flags.cmake for CMAKE_EXE_LINKER_FLAGS_FLEXSPI_NOR_DEBUG
```
    -Xlinker \
    --undefined=qspiflash_config \
    -Xlinker \
    --undefined=image_vector_table \
    -Xlinker \
    --undefined=boot_data \
```
2. Using EXTERN() in linker script to tell linker to find these extern symbols.
```
EXTERN(qspiflash_config)
EXTERN(image_vector_table)
EXTERN(boot_data)
```
3.  Putting XIP related these files directly in application source. 
I will check with our developers to see which approach will be used.
 
Thanks again to your information and sharing of the resolution. Appreciate for your patience and kind input.

mcuxsusan, 2021-08-04:
Hi @hihigupt, during the discussion with my colleague on which resolution we shall use for SDK, he found that there's GCC linking syntax to avoid stripping object files from static library archive. So we would like to suggest you to use `--whole-archive` linker option to use mcux-sdk library to avoid similar issue. You could check [this stackoverflow thread](https://stackoverflow.com/questions/14116420/how-to-force-gcc-to-link-an-unused-static-library) for more details, I have did a quick local test with the linker option `--whole-archive` and see the boot_hdr sections are added in the executable output map file.

mcuxsusan, 2021-08-09:
Hi @hihigupt, have you tried the `--whole-archive` option? Let me know if you have any concern.

hihigupt, 2021-08-09:
Hi @mcuxsusan  it works for me but only one thing based on comments.

—whole-archive to build executables is probably not what you want (due to linking in unneeded code and creating bloated software).
https://stackoverflow.com/questions/805555/ld-linker-question-the-whole-archive-option

Will this cause code bloat and link unnecessary/dead code? I tried to check by including other component from sdk and link it with my application, but I was not getting size increase (I am building MinSize, with gc-sections). I want to make sure if there is no implication on size if we use this flag?

mcuxsusan, 2021-08-11:
Hi @hihigupt, in my local test, no size increase for the executable linking library with `--whole-archive` option. 

I understand your concern that using `-whole-archive` may include unneeded code and data, but this could be eliminated by proper linker flag. Linker could eliminate unused code an data from final executable with option '--gc-sections', to achieve the goal option `-ffunction-sections` and ` -fdata-sections` when compiling objects. Check link at https://gcc.gnu.org/onlinedocs/gnat_ugn/Compilation-options.html for the option description. So, my understanding is that using option `-ffunction-sections` and ` -fdata-sections` for library generation, using option '--gc-sections', `-ffunction-sections` and ` -fdata-sections`  for executable creation could avoid linking unused code and data in final executable even we are using '--whole-archive'.

I also checked the case that whether same symbol in executable would be overwritten by that in the linked library when using '-whole-archive', and the result shows that same symbol will not be overwritten.

danewalton, 2021-08-13:
Closing since it looks like we're unblocked. Thanks for the help!

closed 2021-08-13 by danewalton