nxp-mcuxpresso/legacy-mcux-sdk #31: Freertos blocking state.

Hadatko, 2021-07-22:
Hi

**Is your enhancement proposal related to a problem? Please describe.**
Generaly it is sad that drivers blocking function doesn't count with RTOS. Usualy i saw send/receive blocking functions using while(){}. For example Freertos may use semaphore/event/mutex... as blocking feature so task will not block others tasks while it is waiting for message to be send to other side or from other side.

**Describe the solution you'd like**
It would be nice if blocking calls generally will have implementation for RTOS you are supporting. Otherway everybody who is using RTOS need implements blocking call on its own.

**Describe alternatives you've considered**
-

**Additional context**
For example uart, mu, ...

mcuxsusan, 2021-07-27:
Hi @Hadatko, thanks for reporting. I will check with the team to see if the feature will be planned for future release.

PetrLukasNXP, 2023-05-31:
Hi, this feature is long time supported for some drivers for example : [fsl_spi_freertos.c](https://github.com/nxp-mcuxpresso/mcux-sdk/blob/main/drivers/spi/fsl_spi_freertos.c) . There is no plan to extend this support to other drivers. Drivers which are a part of MCUXpresso SDK comes in both blocking and non-blocking(ISR) variants so they can be easily integrated with RTOS.

Hadatko, 2023-07-02:
Thank you for your statement.

closed 2023-07-02 by Hadatko