nxp-mcuxpresso/legacy-mcux-sdk #33: [BUG] cannot find CMSIS_Include_common [2.10.0]

HGNogueira, 2021-08-02:
**Describe the bug**
The path to "CMSIS_Include_common.cmake" (under CMSIS/Include) is no longer automatically added to the `CMAKE_MODULE_PATH` variable in "all_lib_device_<MCUX_DEVICE>.cmake". This used to be the case in SDK v2.9.0.

Results in the following error when configuring a CMake project that uses mcux-sdk:

```
include could not find requested file:

    CMSIS_Include_common
 ```

**To Reproduce**
* Environment (please complete the following information):
   - Tag/Commit hash: MCUX_2.10.0
   - Toolchain: ARMGCC 9-2020-q2-update
   - Board/SoC: MIMXRT1062
 
* Steps to reproduce the behavior:
Simply try to include the "CMSIS_Include_common" cmake module in your CMake project (using MCUX_SDK). Example:

```
set(MCUX_SDK_PROJECT_NAME <target-executable-name>)
set(MCUX_DEVICE MIMXRT1062)
list(APPEND CMAKE_MODULE_PATH <path-to-all_devices.cmake>)

include(all_devices)
include(CMSIS_Include_common)
# ...
```

**Additional context**
This can be fixed by appending <path-to-CMSIS/Include> to the `CMAKE_MODULE_PATH` under the file included by "all_devices.cmake"

mcuxsusan, 2021-08-03:
Hi @HGNogueira, thanks for the detailed information. I suggest to update your CMakeList.txt to include(CMSIS_Include_core_cm) instead of previous include(CMSIS_Include_common), include(CMSIS_Include_cmxx). Here I use cmxx to refer to all core type, such as cm0plus, cm4, cm7, cm33.
Since MCUX_2.10.0 release, the CMSIS_Include_common.cmake, CMSIS_Include_cmxx.cmake which are located under CMSIS/Include is no longer used, instead a new component CMSIS_Include_core_cm is added, the component cmake file is located under path CMSIS/Core/Include. So, device in [MCUX_2.10.0 release scope](https://github.com/NXPmicro/mcux-sdk/releases/tag/MCUX_2.10.0) all get this update in the "all_lib_device_<MCUX_DEVICE>.cmake"  file, previous CMSIS/Include path is replaced with CMSIS/Core/Include, that's why you see the issue include(CMSIS_Include_common) no longer works. The reason for this change is that we would like to align the component definition for ARM CMSIS pack, only one CMSIS_Include_core_cm component covers different core type. Below provides CMSIS core component definition in pdsc file for CMSIS PACK.
```
    <!-- CMSIS-Core component -->
    <component Cclass="CMSIS" Cgroup="CORE" Cversion="5.3.0"  condition="ARMv6_7_8-M Device" >
      <description>CMSIS-CORE for Cortex-M, SC000, SC300, ARMv8-M, ARMv8.1-M</description>
      <files>
        <!-- CPU independent -->
        <file category="doc"     name="CMSIS/Documentation/Core/html/index.html"/>
        <file category="include" name="CMSIS/Core/Include/"/>
        <file category="header"  name="CMSIS/Core/Include/tz_context.h" condition="ARMv8-M TZ Device"/>
        <!-- Code template -->
        <file category="sourceC" attr="template" condition="ARMv8-M TZ Device" name="CMSIS/Core/Template/ARMv8-M/main_s.c"     version="1.1.1" select="Secure mode 'main' module for ARMv8-M"/>
        <file category="sourceC" attr="template" condition="ARMv8-M TZ Device" name="CMSIS/Core/Template/ARMv8-M/tz_context.c" version="1.1.1" select="RTOS Context Management (TrustZone for ARMv8-M)" />
      </files>
    </component>
```

HGNogueira, 2021-08-03:
Hi @mcuxsusan, thanks for the quick reply!

Looking back at my original post I now realize I've left out an important piece of information:  that I'm using the arm CMSIS DSP library, and it used to be that the include paths were added with [CMSIS_Include_common.cmake](https://github.com/NXPmicro/mcux-sdk/blob/main/CMSIS/Include/CMSIS_Include_common.cmake).

The [CMSIS_Include_core_cm.cmake](https://github.com/NXPmicro/mcux-sdk/blob/main/CMSIS/Core/Include/CMSIS_Include_core_cm.cmake) file only adds the include paths for CMSIS Core. This means that, if I use `include(CMSIS_Include_core_cm)`, I will get the following error: `fatal error: arm_math.h: No such file or directory`.

So, I suppose the question really is: **how does mcux-sdk sees using arm's CMSIS DSP libraries**. Should I add them as separate third-party libraries, independent of mcux-sdk?

mcuxsusan, 2021-08-03:
Hi @HGNogueira, the update information is very helpful. The header arm_math.h belongs to CMSIS DSP library header, for your question, if I understand clearly, you are expecting mcux-sdk includes the CMSIS DSP library delivery, am I right?  Actually the CMSIS DSP should be included in MCUX_2.10.0 release, but I just found there's a bug causing the CMSIS DSP library not be properly added. Working on the fix, the CMSIS DSP library should be available days later. For current solution, you could use the original include(CMSIS_Include_common) by adding module path CMSIS/Include in "all_lib_device_<MCUX_DEVICE>.cmake" file and link external CMSIS DSP library.

HGNogueira, 2021-08-03:
_"if I understand clearly, you are expecting mcux-sdk includes the CMSIS DSP library delivery, am I right?"_
Correct.

Thank you, I will use the workaround for now!

mcuxsusan, 2021-08-04:
Hi @HGNogueira , CMSIS DSP library is now available on latest main branch.  The CMSIS_DSP_Library component need to be included if you would like to use the DSP library. The all_lib_device_MIMXRT1062.cmake file on latest main branch now already added module path for CMSIS_DSP_Library, you could use include(CMSIS_DSP_Library) to include the component. An example CMakeList.txt using the CMSIS_DSP_Library component and also links CMSIS DSP library could be found at link https://github.com/NXPmicro/mcux-sdk-examples/blob/main/evkmimxrt1015/demo_apps/sai/armgcc/CMakeLists.txt.

HGNogueira, 2021-08-04:
Hi @mcuxsusan,

I've applied the recommended changes, with the latest HEAD of the main branch.  It works!

Thank you!

closed 2021-08-04 by HGNogueira