nxp-mcuxpresso/legacy-mcux-sdk #35: [BUG] GCC assembler flags produce warnings when cmake is used.

Hadatko, 2021-08-16:
**Describe the bug**
Does assembler compiler needs so many flags you are providing in CmakeList.txt? 
[image]

I was searching trought internet and it looks like these compiler flags are more or less C specific flags and not assembler. Actually i am not sure if any of mentioned above is usefull maybe only DNDEBUG or DDEBUG. This is related to all examples of all projects. I guess assembler should contains only application specific macros as there is none compiler specific (if i am wrong, correct me)

Why is this issue to me? For example combination of `ENABLE_LANGUAGE(ASM)` and `-std=gnu99`:
https://github.com/microsoft/vscode-cmake-tools/issues/2044#issuecomment-896991481
It would be better to get rid of useless compiler macros for assembler.
So many not used flags for assembler target has also another side-effect  data "noise" in file (file with flags can be smaller and more transparent)

**To Reproduce**
Look into assembler flags for armgcc

**Expected behavior**
Smaller file with flags for compiler. More transparent

Hadatko, 2021-08-16:
Actually i printed `/arm-none-eabi-as --help`
So from asm flags above it looks like these are not related to asm compiler:
* -fno-common
* -ffunction-sections
* -fdata-sections
* -ffreestanding
* -fno-built
* -std=gnu99 (i have warning because this one is used)

`mapcs` looks like incompletly flag.

[asm.txt](https://github.com/NXPmicro/mcux-sdk/files/6991518/asm.txt)

mcuxsusan, 2021-08-18:
Hi @Hadatko , thanks for reporting the issue. Will try to reproduce the issue and see how shall we resolve the incorrect ASM flags. For `-mapcs`, I agree this is incomplete flag, will check with team. Appreciate for patience.

Hadatko, 2021-08-18:
I found this note related to arm options 3.19.5
[image]

mcuxsusan, 2021-08-18:
@Hadatko, thanks for clarification on why `-maps` now seems an incomplete flag.

mcuxsusan, 2021-08-19:
Hi @Hadatko, I reproduced the warning "[cpptools] Unknown language: ASM" with cmake-tools in vscode. And you are correct, it's caused by the incorrect ASM flag '-std=gnu99', we will fix it in next release. For below flags you listed, currently in our implementation they are parsing to arm-none-eabi-gcc to assemble ASM files, not parse to the arm-none-eabi-as. I am not sure whether the options received via gcc will directly parse to the assembler, but per my test, these options could be removed because there's no difference in map file and elf file.
* -fno-common
* -ffunction-sections
* -fdata-sections
* -ffreestanding
* -fno-built
* -mapcs

Hadatko, 2021-08-19:
I guess there was reason for -mapcs in past. Did you tested also -mapcs-frame? 
Thank you very much to solving this. I will close issue as you are going to solve it internally.

closed 2021-08-19 by Hadatko

mcuxsusan, 2021-08-19:
Yes, I tested also -mapcs-frame, for ASM flags, this should be removed because it does not have effect either map or elf, could not removed. For C flags, the generated elf and output map shows difference for debug_str location, the binary is same.