nxp-mcuxpresso/legacy-mcux-sdk #36: fsl_i2c.c: Clearing the NAK flag prevents scan.

robert-hh, 2021-08-18:
The following code lines (1068ff) clear the NAK flag on the last byte before checking. 
```
    /* For the last byte, nack flag is expected.
       Do not check and clear kLPI2C_MasterNackDetectFlag for the last byte,
       in case FIFO is emptied when stop command has not been sent. */
    if (handle->remainingBytes == 0U)
    {
        status &= ~(uint32_t)kLPI2C_MasterNackDetectFlag;
    }
```
That interferes with an attempt to scan the I2C bus for addresses, when just a single byte is sent and the NAK flag returns essential information. This part of the code sneaking in between 2019 and 2020. It works wirth the 2019 code and if I simply drop this check. So what was the reason to introduce it, and for that case, is is feasible to do that check only for messages of length > 1?

mcuxsusan, 2021-08-19:
Thanks for raising the issue, will check with developer and provide feedback later.

robert-hh, 2021-08-19:
Hello Susan,
thank you for the swift reply. On a somewhat related note: Do you know where I can get the schematics of the MIMXRT1050_EVKB bord. It is SCH-30168 REV A1. The ones available at the NXP site are for REV B1,and these do not match the board in all details. Or if they should, they have errors.
Best Regards, Robert

mcuxsusan, 2021-08-19:
Hi @robert-hh, I searched the SCH-30168 REV A1 with Google and see the documentation download link is available in our NXP community, you could download from link https://community.nxp.com/pwmxy87654/attachments/pwmxy87654/imx-processors/152996/1/SPF-30168_A1.pdf and check if this is the one you are searching for.

On NXP site we only provide latest schematic version, so the latest board version for MIMXRT1050_EVKB is REV B1, appreciate for your understanding.

robert-hh, 2021-08-19:
Thanks a lot. That is the right one. I looked for it with google as well and did not get the link. Maybe it's not that good to provide information only for the latest board, when the other boards are still on sale and even longer in the field. I bough my board 3 weeks ago at DigiKey.

mcuxsusan, 2021-08-23:
Hi @robert-hh, I asked the team and the feedback is that the policy on our website is to release the latest documentation. If in future you have similar issue that specific documentation could not be found, you may post a question in our NXP community https://community.nxp.com/t5/i-MX-RT/bd-p/imxrt for feedback, this could help.

For clearing NAK flag of master check on the last byte, the scenario is that there's situation that slave may return NAK on last byte of transfer but it's a successful transfer, and this does not relates to the transfer length. For such scenario, if we did not clear NAK status for the check, the successful transfer will return status NAK which may makes user confuse, so we just simply ignore the NAK flag on last byte. For your case, seems the information is needed. We have similar issue reported on another I2C driver that there's case the NAK flag on last byte need to be checked, and the use case is as below:

The used i2c device that does not support restart signal, so the read slave register transfer divides to two separate transfer: The first transfer write the device register, and the second register read slave.  For the first transfer, if the write register is not ready, device will return NAK so the return status need to be checked.

For this requirement, the developer introduces a new macro in the driver to decide whether the NAK flag on last byte need to be checked. Are you expecting similar feature?

robert-hh, 2021-08-23:
> For this requirement, the developer introduces a new macro in the driver to decide whether the NAK flag on last byte need to be checked. Are you expecting similar feature?

My case is, that the NAK flag is used to tell, whether a device with a certain address exists on the bus.  In order to probe the address, the master simply send a single address byte. In case of no response the master reads a NAK status. So if that NAK is suppressed, the master assumes that there is a device with that address on the bus. At the moment I added a fix, that this NAK suppression only takes place for transfers longer that 1 byte. I cannot change this probing mechanism to say a 2 byte transfer, because it is set up as such for many different MCU types and vendors, and the actual NXP lib seems to be the only one with this unconventional behavior.
Interestingly, this NAK supressing only happens when using non-blocking transfer. Using blocking-transfer works well.

mcuxsusan, 2021-08-23:
> My case is, that the NAK flag is used to tell, whether a device with a certain address exists on the bus. In order to probe the address, the master simply send a single address byte. In case of no [...]

Thanks for the update information. So, if my understanding is correct, your case is that you make up transfers which contains device address and one byte of data for device probe, but the probe transfer to non-exist device does not return NAK flag as expected. If so, this seems to be an bug in transfer.

robert-hh, 2021-08-23:
> So, if my understanding is correct, your case is that you make up transfers which contains device address and one byte of data for device probe,

No. The probe consists of a one byte transfer consisting of the address byte only, which already according to the standard must return NAK if the device does not exist. If you send a single byte only, that byte is also the last one in the transfer.

mcuxsusan, 2021-08-23:
> No. The probe consists of a one byte transfer consisting of the address byte only, which already according to the standard must return NAK if the device does not exist. If you send a single byte [...]

Thanks for providing the detail. The information is very helpful for owner to reproduce the case. It's quite a common case to do device probe on I2C bus, will provide feedback later.

mcuxsusan, 2021-09-03:
Hi @robert-hh, excuse for my late update. Our developer has reproduced the bug and provided fix in https://github.com/NXPmicro/mcux-sdk/commit/e84ccf99a203bbbd8872d9953c8d3b57864d484b, you could apply the fix and see if it works.

robert-hh, 2021-09-03:
Hello @mcuxsusan, thank you for the feedback. Do not worry about the time. I had a workaround. I will test the fix in the next days. The source code looks well.

robert-hh, 2021-09-03:
Hello @mcuxsusan. I found some time to run tests with the changed files. It looks fine. Everything I tried worked as expected. Thank you very much again for your swift reaction. 
Kind Regards, Robert

closed 2021-09-03 by robert-hh