nxp-mcuxpresso/legacy-mcux-sdk #39: Should standard function return -1 when implementation is disabled?

Hadatko, 2021-09-08:
Hello, sorry for bothering you. I saw this code change 
https://github.com/NXPmicro/mcux-sdk/blob/e84ccf99a203bbbd8872d9953c8d3b57864d484b/utilities/debug_console/debug_console/fsl_debug_console.h#L62

And i wanted to ask (may give other perspective) why the functions PRINTF, SCANF, ... are returning -1 when '#if SDK_DEBUGCONSOLE == DEBUGCONSOLE_DISABLE' ? I would maybe expect that it will return 0 as it is disabled and expected that nothing will happen. Negative numbers are for errors.  I see that current impementation has closer to previous one (infinite loop).

mcuxsusan, 2021-09-08:
Hi @Hadatko, no worry for disturb, always welcome for your suggestion and contribution.
In my opinion, when SDK_DEBUGCONSOLE is defined as DEBUGCONSOLE_DISABLE, the behavior of calling to function PRINTF/SCANF/PUTCHAR/GETCHAR is not expected. So, it returns -1 to show there's error usage for these functions.

Adding @lylezhu2014 for clarification.

Hadatko, 2021-09-08:
i think this doesn't have simple answer. Because it depends on usecases. For example one example could be simple remove for PRINTF calls from code without changing  logic. Logic could be:
```C
int c = PRINTF("text");
if (c<0)
{
//error happend
}
else if(c == 0)
{
//nothing to print or print is disabled
}
else
{
// chars are printed
}
```
With current solution it would looks like
```C
int c = PRINTF("text");
if (c<0)
{
//error happend
}
#if SDK_DEBUGCONSOLE == DEBUGCONSOLE_DISABLE
else if(c == 0)
{
//print is disabled
}
#else
else if(c == 0)
{
//nothing to print
}
#endif
else
{
 //chars are printed
}
```

Anyway thank you for explanation.

closed 2021-09-08 by Hadatko