nxp-mcuxpresso/legacy-mcux-sdk #40: Simplify include <stdio.h>

Hadatko, 2021-09-08:
This condition can be theoreticaly be removed
https://github.com/NXPmicro/mcux-sdk/blob/e84ccf99a203bbbd8872d9953c8d3b57864d484b/utilities/debug_console/debug_console/fsl_debug_console.h#L51
 and include can be moved under
https://github.com/NXPmicro/mcux-sdk/blob/e84ccf99a203bbbd8872d9953c8d3b57864d484b/utilities/debug_console/debug_console/fsl_debug_console.h#L76
or am i wrong?

mcuxsusan, 2021-09-08:
Hi @lylezhu2014, could you help explain?

Hadatko, 2021-12-15:
[image]
[image]

mcuxsusan, 2021-12-15:
Hi @Hadatko, thanks for your patience. @lylezhu2014, please help update your progress here.

lylezhu2014, 2021-12-17:
> [image] [image]

Hi @Hadatko , I don't see the difference between this change and the previous one. Does this change have any special effect for you?

Hadatko, 2022-07-25:
Less condition more transparent? And also redundant include for disabled debug console?

bdcncdcdvd, 2023-07-24:
Hi, @Hadatko. I don't think it is more transparent if make some changes. Because now the "fsl_debug_console.h" has been changed as follow:
[image]
As we can see, there are one condition be added, if we update the code as you said then the follow codes will be redundant：
[image]
[image]

mcuxsusan, 2023-07-24:
It seems the header file only needs to be included under some conditions, that's design intention, so I am gonna to close the issue.

closed 2023-07-24 by mcuxsusan