nxp-mcuxpresso/legacy-mcux-sdk #51: LPSPI RTOS driver needs a completion callback

andresovela, 2021-11-09:
**Is your enhancement proposal related to a problem? Please describe.**
The API `fsl_lpspi_freertos` component does not allow the user to provide a completion callback. Such APIs already exist for the `fsl_lpspi` and `fsl_lpspi_edma` components

See:
- `LPSPI_MasterTransferCreateHandle()` in `fsl_lpspi.h`
- `LPSPI_MasterTransferCreateHandleEDMA()` in `fsl_lpspi_edma.h`

**Describe the solution you'd like**
A similar API for `fsl_lpspi_freertos` that allows the user to provide a completion callback.

**Describe alternatives you've considered**
- Adding a function to check if the transfer is completed by polling, checking if the event semaphore is still taken (I'd say this is a no-go).

**Additional context**
I need this API to do some other work in the same FreeRTOS task, rather than blocking the task until the mutex is released by the SPI driver.

mcuxsusan, 2021-11-10:
@andresovela, thanks for the report. Added @DavidJurajdaNXP to help check the improvement.

DavidJurajdaNXP, 2024-02-19:
Hi @andresovela, thank you for comment. At this point we are no longer planning to extend the FreeRTOS API by new functionality. Drivers are currently in maintenance mode only, where we concentrate only on correction of critical issues.