nxp-mcuxpresso/legacy-mcux-sdk #52: MIMXRT685 missing dsp header files

mmahadevan108, 2021-11-11:
**Is your enhancement proposal related to a problem? Please describe.**
The `/devices/MIMXRT685S` folder is missing the `_dsp` header files: `MIMXRT685S_dsp.h` & `MIMXRT685S_dsp_features.h`

**Describe the solution you'd like**
Please add files `MIMXRT685S_dsp.h` & `MIMXRT685S_dsp_features.h` to `/devices/MIMXRT685S` folder

mmahadevan108, 2021-11-11:
cc @mcuxsusan

suchun230, 2021-11-14:
Hi @mmahadevan108, Added in latest main branch.
I checked with the team and found these dsp header file/feature files recorded in repo have dependency to xcc, xtensa toolchain. Since currently we only support ARMGCC and MCUX so these files are not added. Now add them for your needs. It would be great if you could share how zephyr uses these header files so I could ask the team to remove the file dependency to toolchain.

mcuxsusan, 2021-11-14:
Hi @mmahadevan108, Added in latest main branch.
I checked with the team and found these dsp header file/feature files recorded in repo have dependency to xcc, xtensa toolchain. Since currently we only support ARMGCC and MCUX so these files are not added. Now add them for your needs. It would be great if you could share how zephyr uses these header files so I could ask the team to remove the file dependency to toolchain.

closed 2021-11-14 by mcuxsusan

mmahadevan108, 2021-11-15:
These files were included in the MXRT685 release and we added these to Zephyr. These might be needed for future work to add DSP core support under Zephyr.