nxp-mcuxpresso/legacy-mcux-sdk #57: Certain devices folder are missing .cmake files

mmahadevan108, 2021-12-15:
**Is your enhancement proposal related to a problem? Please describe.**
Below list of device are missing `.cmake` files. This will cause build issues. This was highlighted by a Zephyr customer who was trying to use `RT1051` which is missing the .cmake files. I compiled a complete list of `devices` that are missing these `.cmake` files.
```
K32L2A31A/
K32L2B11A/
K32L2B21A/
LPC54005/
LPC54016/
LPC54018/
LPC54018M/
LPC54113/
LPC54605/
LPC54606/
LPC54607/
LPC54608/
LPC54616/
LPC54618/
LPC54S005/
LPC54S016/
LPC5502/
LPC5502CPXXXX/
LPC5504/
LPC5504CPXXXX/
LPC5512/
LPC5514/
LPC5516/
LPC5526/
LPC5528/
LPC55S04/
LPC55S14/
LPC55S26/
LPC55S66/
LPC810/
LPC811/
LPC822/
LPC832/
LPC834/
LPC844/
MCIMX7U3/
MCIMX7U5/
MIMX8DX1/
MIMX8DX2/
MIMX8DX3/
MIMX8DX4/
MIMX8DX5/
MIMX8DX6/
MIMX8MD6/
MIMX8MD7/
MIMX8ML3/
MIMX8ML4/
MIMX8ML6/
MIMX8MM1/
MIMX8MM2/
MIMX8MM3/
MIMX8MM4/
MIMX8MM5/
MIMX8MN1/
MIMX8MN2/
MIMX8MN3/
MIMX8MN4/
MIMX8MN5/
MIMX8MQ5/
MIMX8MQ7/
MIMX8QX1/
MIMX8QX2/
MIMX8QX3/
MIMX8QX4/
MIMX8QX5/
MIMX8UX5/
MIMX8UX6/
MIMXRT1051/
MIMXRT1061/
MIMXRT1165/
MIMXRT1171/
MIMXRT1172/
MIMXRT1173/
MIMXRT1175/
MIMXRT633S/
MK02F12810/
MK22F12810/
MK22F25612/
MK24F12/
MK26F18/
MK27FA15/
MK63F12/
MK65F18/
MK80F25615/
MK82F25615/
MKE04Z1284/
MKE14F16/
MKE14Z4/
MKE14Z7/
MKE15Z4/
MKE16F16/
MKL17Z644/
MKL25Z4/
MKM14ZA5/
MKM33ZA5/
MKV10Z1287/
MKV10Z7/
MKV30F12810/
MKV31F12810/
MKV31F25612/
MKV56F24/
MKW20Z4/
MKW21Z4/
MKW22D5/
MKW24D5/
MKW30Z4/
MKW31Z4/
MKW40Z4/
MKW41Z4/
```

**Describe the solution you'd like**
Please add `.cmake` files similar to other platforms

mmahadevan108, 2021-12-15:
@mcuxsusan @dleach02

mcuxsusan, 2021-12-17:
Hello, the reason why you do not see device cmake files in these folder is because they are subset devices. Currently we only provide cmake component for main device support. I have created task to provide cmake component for subset devices, will update the status once the task is done.

alonbl, 2024-10-03:
On  Dec 17, 2021:
> I have created task to provide cmake component for subset devices, will update the status once the task is done.

Any news?