nxp-mcuxpresso/legacy-mcux-sdk #58: Aligning lpuart config and rtos lpuart config names.

Hadatko, 2021-12-20:
**Is your enhancement proposal related to a problem? Please describe.**

Hello, is there reason why lpuart_config and lpuart_rtos_config have different names for theirs members which has same functionality? It looks a bit messy to me. Maybe both configuration may have common structure which they may use inside. 

https://github.com/NXPmicro/mcux-sdk/blob/81a42c9e5875b70f0acdf10c23aa977ce1135100/drivers/lpuart/fsl_lpuart.h#L240
https://github.com/NXPmicro/mcux-sdk/blob/81a42c9e5875b70f0acdf10c23aa977ce1135100/drivers/lpuart/fsl_lpuart_freertos.h#L32

**Describe the solution you'd like**
Align members names or create new common structure. 

**Describe alternatives you've considered**
Do nothing

**Additional context**

mcuxsusan, 2021-12-20:
@Hadatko, thanks for your suggestion. The developer @DavidJurajdaNXP is currently on holiday leave so the feedback could be delayed. Appreciate for your patience and also wish you happy holiday.

Hadatko, 2021-12-20:
No problem, thank you. I would wish you same but i think you personali don't have holidays right now correct?

mcuxsusan, 2021-12-20:
> No problem, thank you. I would wish you same but i think you personali don't have holidays right now correct?

Yes, you are correct. Our holiday for family to get together will be about a month later, thank you all the same.

DavidJurajdaNXP, 2024-02-22:
Hi @Hadatko, thank you for your question. Misalignment of member names is caused by that each layer was created by different author. At this point we are no longer planning to extend the FreeRTOS API by new functionality. FreeRTOS drivers are currently in maintenance mode only, where we concentrate only on correction of critical issues.

closed 2024-02-22 by DavidJurajdaNXP