nxp-mcuxpresso/legacy-mcux-sdk #59: Any specific reason why rtos lpuart send function data buffer variable is not const?

Hadatko, 2021-12-20:
https://github.com/NXPmicro/mcux-sdk/blob/81a42c9e5875b70f0acdf10c23aa977ce1135100/drivers/lpuart/fsl_lpuart_freertos.h#L132

In Send function the data shouldn't be changed and it looks like they are not. Isn't better to use const variable type? (i have data type conflict because it is not using const keyword)

Hadatko, 2022-12-05:
?

DRNadler, 2023-05-23:
NXP, Anybody there? I also have lots of problems with FSL API function arguments that should be const. Its a big application problem when sending lots of const configuration strings from flash, etc. Thanks!

mcuxsusan, 2023-05-24:
Hi @DRNadler, appreciate if you could raise a new issue and provide more details for your case, I will communicate with develop team for feedback.

mcuxsusan, 2023-05-24:
> 

Sorry for the delayed reply. @DavidJurajdaNXP, could you please help check?

escherstair, 2023-12-11:
Hi @mcuxsusan , @DavidJurajdaNXP 
I add my vote to this request.
Anmd I point your attention to `fsl_uart_freertos.h` too
https://github.com/nxp-mcuxpresso/mcux-sdk/blob/43a5a8df5728436f88dd1044fba66972bae723b6/drivers/uart/fsl_uart_freertos.h#L126

escherstair, 2024-02-12:
@mcuxsusan , @DavidJurajdaNXP 
sorry for pushing, but do you have news on this issue?
Does my PR sound good to you?

DavidJurajdaNXP, 2024-02-20:
Hi all, there is not any specific reason for missing const modifier, it is design choice of original author. At this point we are no longer planning to extend the FreeRTOS API by new functionality. Drivers are currently in maintenance mode only, where we concentrate only on correction of critical issues.

closed 2024-02-20 by DavidJurajdaNXP