nxp-mcuxpresso/legacy-mcux-sdk #68: wm6890.c: Suspicious bit rate calculation.

robert-hh, 2022-02-02:
https://github.com/NXPmicro/mcux-sdk/blob/7b545788e3f6e0c1a8decae21e428ed8d5b5a8e7/components/codec/wm8960/fsl_wm8960.c#L103

This line looks strange, since it has the '* 2U' term in both numerator and denominator. In fact that function creates a bit rate which is half of what is should be. While recording from a microphone with a sample rate of 16000 and 16 bit width, codec set to master mode, the bit rate was 512kHz. It should have been 1024 kHz. Dropping the '* 2U' in the numerator fixed it.

mcuxcc, 2022-02-16:
Hi @robert-hh 

Thank you for your feedback.

I am sorry for late reply.

Let us discuss the issue you report,

The *2U added in the numerator is used to expand the bit clock divider to an integer, as the calculated bit clock divider maybe a fractional value
[image]
So *2U is added for non-float calculation.

robert-hh, 2022-02-16:
I understand the intention. However, using that code I got a wrong bitclock frequency in master mode when recording from a microphone. The frequency was half of what it should be. So maybe the problem is somewhere else.

mcuxcc, 2022-02-16:
is the issue come from the ***2** of denominator?

robert-hh, 2022-02-16:
Even in Mono mode, both channels are present. For Mono, the signal is taken from the left channel. And, there is a second mic input at the headphone connector of the Dev boards.

If the divisor is smaller, the resulting frequency is higher.

robert-hh, 2022-02-17:
I compared that with the timing in slave mode, and it seems fine.

closed 2022-02-17 by robert-hh