nxp-mcuxpresso/legacy-mcux-sdk #69: wm8960.c: Precision loss in calculation.

robert-hh, 2022-02-08:
https://github.com/NXPmicro/mcux-sdk/blob/7b545788e3f6e0c1a8decae21e428ed8d5b5a8e7/components/codec/wm8960/fsl_wm8960.c#L68

The order of arguments in this expression causes wrong values for pllN. For example with iniputMclk of 24MHz and outputClk of 11.2896MHz, the calculated pllN is 6, when it should be 7. If sticking with integer arithmetic, the line must read;

```
            pllN      = (pllF2 * sysclkDiv) / inputMclk;
```

I added brackets to make clear, what must happen. The same happens with fewer problems a few lines down when calculating pllR. There, the error is 1000 times smaller.

mcuxcc, 2022-02-16:
Hi @robert-hh 

Thank you for the feedback.

I am sorry for late the reply. 

I am agree with you about the issue, so i will create a Pull request for it.

Thanks!

closed 2022-02-17 by robert-hh

mcuxsusan, 2022-03-14:
Update: the improvement patch is available in https://github.com/NXPmicro/mcux-sdk/commit/ce9a8be7b39ac939fba6a4527a079ae98eb045b2

robert-hh, 2022-03-14:
Thank you very much. Just for information: I use a version of the driver translated to Python. It is here: https://github.com/robert-hh/wm8960