nxp-mcuxpresso/legacy-mcux-sdk #73: video/camera/receiver/isi/fsl_isi_camera_adapter component should support output frame with raw8/raw10 format.

SimonFu0506, 2022-02-28:
**Is your enhancement proposal related to a problem? Please describe.**
There are some camera only support raw8/raw10 format. So it is batter that video/camera/receiver/isi/fsl_isi_camera_adapter component should support output frame with raw8/raw10 format. 

**Describe the solution you'd like**
video/camera/receiver/isi/fsl_isi_camera_adapter component can output the raw8/raw10 format.

zejiang0jason, 2022-02-28:
Hi @SimonFu0506 , thanks for feedback. There is RAW8 support, the example is here:
https://github.com/NXPmicro/mcux-sdk-examples/blob/main/evkbimxrt1050/driver_examples/csi/raw8/csi_raw8.c

Would you please check whether it meets the RAW8 requirement? If so, RAW10 can be added similarly.

Thanks.

SimonFu0506, 2022-03-01:
Hi @zejiang0jason,

Thanks for your response.  I have different hw with the example.  Following is my board and MCU.
**Board: mekmimx8qm**
**Devices: MIMX8QM6**
**Camera:  OV7251 (Frame is Raw8)**

My MCU supports the ISI module. So the [camera receiver](https://github.com/NXPmicro/mcux-sdk/blob/main/components/video/camera/receiver/isi/fsl_isi_camera_adapter.c) is based on ISI module. In the camera receiver module, there is a map that covert the video format to ISI format.

```
 static const isi_output_format_map_t s_isiOutputFormatMap[] = {{
                                                                   .fourccFormat = kVIDEO_PixelFormatXRGB8888,
                                                                   .isiFormat    = kISI_OutputXRGB8888,
                                                               },
                                                               {
                                                                   .fourccFormat = kVIDEO_PixelFormatRGB888,
                                                                   .isiFormat    = kISI_OutputRGB888,
                                                               },
                                                               {
                                                                   .fourccFormat = kVIDEO_PixelFormatRGB565,
                                                                   .isiFormat    = kISI_OutputRGB565,
                                                               },
                                                               {
                                                                   .fourccFormat = kVIDEO_PixelFormatYUYV,
                                                                   .isiFormat    = kISI_OutputYUV422_1P8P,
                                                               }};
```

But there is no item for raw8. In order to support raw8 format, I add a item in the map.
`                                                             {
                                                                   .fourccFormat = kVIDEO_PixelFormatRAW8,
                                                                   .isiFormat = kISI_OutputRaw8,
                                                               },`
 
It almost work. But I got attached frames of color bar test pattern. 
[image]

The frames only have 4 gray levels and lost most of the information. Could you help check the SDK for my hw platform?

Thanks
Simon

zejiang0jason, 2022-03-02:
Hi @SimonFu0506 ,

Please try to modify the function MIPI_CSI2RX_InitInterface in file fsl_mipi_csi2rx.c, change `csr->DATA_TYPE_DIS       = 0U;`  to `csr->DATA_TYPE_DIS       = 0x80000000U;`.

Thanks.
Jason

zejiang0jason, 2022-03-08:
Hi @SimonFu0506 ,

After confirming internally, there is a system hardware integration issue with MIPI CSI and ISI. For RAW data, the total data bus width from CSI to ISI is 14-bit, and the input data is aligned with MSB. In other word, when you use RAW8 data, the RAW8 data is fed to ISI input bit13~bit6, the ISI input bit5~bit0 is always 0, so in the result, only 2-bit are saved in memory, as you see.

The fix is only applied to imx8qx_c0, with this fix, the offset can be corrected by setting CSR register bit31. IMX8QM can't support the RAW8 data.

Thanks.
Jason

SimonFu0506, 2022-03-09:
Hi @zejiang0jason,

Thanks for your reply. That means MIMX8QM6 cannot support Raw8 frame, right?

How about the raw10 frame? Is it similar with Raw8 frame? 

Besides, I have also tried Raw10 frame. It is incorrect too.  
[image]

Thanks
Simon

mcuxsusan, 2022-03-14:
Closed the issue as the feedback from @zejiang0jason stated that the RAW8 data cannot be supported on IMX8QM.

closed 2022-03-14 by mcuxsusan