nxp-mcuxpresso/legacy-mcux-sdk #78: Register file is not compilable with gcc and C++ (MIMX8ML8_cm7.h but likely others too)

mwadva, 2022-03-17:
**Is your enhancement proposal related to a problem? Please describe.**

we're using MCUXpresso with FreeRTOS on a Variscite Module (imx8mp-var-dart), targeting the Cortex M7 Coprocessor.

While FreeRTOS itself is fine with gcc and C++, we cannot go down that route due to issues with the register file, in our case ```devices/MIMX8ML8/MIMX8ML8_cm7.h```:

```
.../devices/MIMX8ML8/./MIMX8ML8_cm7.h:10358:51: error: 'reinterpret_cast<CCM_Type*>(808976384)' is not a constant expression
10358 | #define CCM                                      ((CCM_Type *)CCM_BASE)
.../devices/MIMX8ML8/drivers/./fsl_clock.h:405:44: note: in expansion of macro 'CCM'
  405 |     kCLOCK_RootM7           = (uint32_t)(&(CCM)->ROOT[1].TARGET_ROOT),  /*!< ARM Cortex-M7 Clock control name.*/
      |                                            ^~~
```

**Describe the solution you'd like**

make the register file compliant with gcc's C++ compiler
```
arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
```

**Describe alternatives you've considered**

none yet

we failed trying to wrap the file with the usual extern C compiler hint

```
#if defined(__cplusplus)
extern "C" {
#endif
```

The boilerplate of that register-file indicates, but is also not clear about other compilers in concunction with C++:

```
6 **     Compilers:           GNU C Compiler
7 **                          IAR ANSI C/C++ Compiler for ARM
8 **                          Keil ARM C/C++ Compiler
```

still it would be nice, wise and smart to make it work with g++

**Additional context**

- NXP MCUXpresso https://github.com/NXPmicro/mcux-sdk Version 2.10.0 
- The C compiler identification is GNU 10.2.1
- The CXX compiler identification is GNU 10.2.1 
- from the gcc-arm-none-eabi-10-2020-q4-major tarball

mcuxsusan, 2022-03-18:
Thanks for the information of, forward to develop team for checking. Reply could  be delayed and appreciate for your patience.

mcuxsusan, 2022-05-06:
Hi, develop team checked the issue and suggested below fix:
changing the enumeration assignment from (uint32_t)(&(CCM)->ROOT[1].TARGET_ROOT) to assignment as below:
(uint32_t)CCM_BASE + offsetof(CCM_Type, ROOT[1].TARGET_ROOT)

The updated patch include other new features so it still under checking. Will try to provide patch this week. Appreciate for your patience.

mcuxsusan, 2022-05-07:
Fix has been applied to the latest main branch, please check commit https://github.com/NXPmicro/mcux-sdk/commit/7d2944c528ba4fe4416c7796a9186a9ac66e8b98

closed 2022-05-07 by mcuxsusan