nxp-mcuxpresso/legacy-mcux-sdk #81: Enhancement: Multi target CMake project support

imi415, 2022-04-12:
Currently, MCUXpresso SDK only supports single target in top level CMakeLists.txt, 
since the components and drivers uses the following method adding source files to the build:

```CMake
target_sources(${MCUX_SDK_PROJECT_NAME} PRIVATE
    #...something belongs to the component...
)

target_include_directories(${MCUX_SDK_PROJECT_NAME} PUBLIC
   #...some directories needs to be searched...
)
```

This method requires there should only be one single target (namely `${MCUX_SDK_PROJECT_NAME}`), adding another target to the same project (with use cases below) will not work without touching the SDK content or adding them manually.

**Typical use cases:**

* Linking to somewhere else by changing linker script: SRAM, FlexSPI, SDRAM..etc, you name it.
* Different build configurations, compiler definitions

**Possible solution:**
It would be nice if components are re-organized as static library targets, and built as dependencies of the main executable target... maybe not the best solution, and may requires a lot of effort...

mcuxsusan, 2022-04-20:
Thanks for reporting the issue. Will need some time for investigation, the team is quite busy so the reply will be delayed. Appreciate for your patience.

tq-steina, 2023-04-20:
A short term solution to this would be integrate #30 as in this case those `.cmake` files will be included again, with `MCUX_SDK_PROJECT_NAME` (hopefully) being set to the new project target.

The long-term solution is to use (alias) targets to object libraries. `drivers/common/driver_common.cmake` might look like this:
```cmake
#Description: Common Driver; user_visible: True
message("driver_common component is included.")

add_library(driver_common OBJECT
    ${CMAKE_CURRENT_LIST_DIR}/fsl_common.c
    ${CMAKE_CURRENT_LIST_DIR}/fsl_common_arm.c
)

target_include_directories(driver_common PUBLIC
    ${CMAKE_CURRENT_LIST_DIR}/.
)

target_link_libraries(driver_common
    core::driver::reset
    core::devices::CMSIS
)
add_library(core::driver::common ALIAS driver_common)
```

This is then referenced using 
```cmake
target_link_libraries(${MCUX_SDK_PROJECT_NAME} PRIVATE core::driver::common)
```

The benefit of aliased targets using `::` is that they **must** be CMake targets. These cannot be fullfilled by some (stray) files. Refer to https://cmake.org/cmake/help/latest/command/target_link_libraries.html

This would look much more like "Modern CMake" where everything is passed using targets and properties and not using variables.

Badokas, 2024-04-13:
+1 We need multi-target CMake support for dual-core NXP MCUs.