nxp-mcuxpresso/legacy-mcux-sdk #82: Eliminate differences between various MCUXpresso SDKs

dsyx, 2022-04-22:
Currently, there are three ways to obtain the MCUXpresso SDK:

* [MCUXpresso SDK Builder](https://mcuxpresso.nxp.com/en/), 'BuilderSDK' for short
* [MCUXpresso IDE](https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/mcuxpresso-integrated-development-environment-ide:MCUXpresso-IDE), 'IDESDK' for short
* [GitHub](https://github.com/NXPmicro/mcux-sdk), 'GitHubSDK' for short

I want to configure a general development environment based on VSCode + CMake + GNU Arm Embedded Toolchain for NXP MCU. However, none of these three SDKs are perfect for me:

* GitHubSDK contains code for various devices, but lacks a lot of middleware compared to the other two SDKs.
* The BuilderSDK can only select a single device or board to build, I may need to configure the workspace environment for each device or board individually.

Is there a way to build a BuilderSDK for multiple devices/boards or to complement the middleware that GitHubSDK is missing?

Thanks

mcuxsusan, 2022-04-22:
Hi @dsyx, unfortunately we do not have the way to get all devices/boards with full middleware support for now. We are still in the way enriching GitHub SDK to put more middleware, but that's not a short-term thing.

dsyx, 2022-04-22:
Hi @mcuxsusan ,
It does take a long time to migrate middleware. So is there a way to make the BuilderSDK support multiple devices/boards?

mcuxsusan, 2022-04-22:
> Hi @mcuxsusan , It does take a long time to migrate middleware. So is there a way to make the BuilderSDK support multiple devices/boards?

Hi, the SDK Builder only provides SDK for single board, because customers mostly only working on single device. I appreciate your effort on creating ecosystem for all users to ease their development on all NXP MCUs, this matches our design of GitHub SDK in which we provide a superset of board/device support.

dsyx, 2022-04-22:
Thanks for your answer.

closed 2022-04-22 by dsyx