nxp-mcuxpresso/legacy-mcux-sdk #89: Include USB RNDIS Ethernet Driver code so we can contribute enhancements

tgibson-work, 2022-07-27:
**Is your enhancement proposal related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

mcuxsusan, 2022-07-27:
Hi @tgibson-work, may you please check the repository https://github.com/NXPmicro/mcux-sdk-middleware-usb?
For NXP GitHub SDK we adopted a multi-repo framework and each middleware is located in their repository. Using zephyr west tool to get https://github.com/NXPmicro/mcux-sdk/ you could get all SDK deliveries.

tgibson-work, 2022-07-27:
Hi @mcuxsusan - thanks for the heads-up, I will check out zephyr west.

However, I have just had a look at the middleware-usb repository and it doesn't include the `usb_ethernetif_freertos.c` `usb_ethernetif.h` files we have modified which are under the `./lwip/port` subfolder in ZIP SDKs (I presume there's a `usb_ethernetif_bm.c` or similar file somewhere as well but we don't use BM at the moment).

I will get all the repositories with zephyr west and see if the files are in another one, but perhaps you might know another repository I should check also? Thanks!

mcuxsusan, 2022-07-27:
Hi @tgibson-work, thanks for the detailed information. Sorry that not all available middleware in MCUX SDK is ready in GitHub now, for the lwip middleware repository which may include the file you could like to contribute is currently not available, the plan is in next month. I will update you once the lwip repository is available.

By the way, for all available middleware list please check the west manifest file: https://github.com/NXPmicro/mcux-sdk/blob/main/west.yml

tgibson-work, 2022-07-27:
@mcuxsusan no problem, great to know the lwip repository is planned for release soon!

mcuxsusan, 2022-08-30:
Hello @tgibson-work, the NXP lwip repository has been launched! And the porting files you are expecting is located under https://github.com/NXPmicro/lwip/tree/mcux_release_2.12.0/port. You could use west tool to do update on our MCUX_2.12.0_UPDATE tag in this repository to checkout the lwip repository.

But for contributions to this lwip repository, please wait notification from @stanislav-poboril who is the repo maintainer. Thanks in advance.

tgibson-work, 2022-09-01:
@mcuxsusan that's great news, thanks for the update!

tgibson-work, 2022-10-06:
> You could use west tool to do update on our MCUX_2.12.0_UPDATE tag in this repository to checkout the lwip repository.

What is the correct procedure for doing this? The mcux-sdk repository documentation covers checking out a new repository to a specified tag, but not how to change the tag and update. How do I do this? I have read through the west documentation and it does not seem to explain how to get a new version of the manifest either.

stanislav-poboril, 2022-10-06:
@tgibson-work, you can navigate to the manifest repository (`core` subfolder of the west workspace) and checkout the desired tag, sha or branch using regular git commands. This will update the manifest file `core/west.yml` which is referenced from the west workspace (configured in the file `.west/config`). The command `west update` will update all references according to the manifest file then. @mcuxsusan, maybe there is a better way using the `west` tool only?

tgibson-work, 2022-10-06:
Thanks, I'll give that a go.