nxp-mcuxpresso/legacy-mcux-sdk #90: [BUG] west.yml references nonexistant mcux-sdk-middleware-ethermind repository

tgibson-work, 2022-07-29:
**Describe the bug**
When checking out mcux-sdk with west it tries to fetch a nonexistant repository mcux-sdk-middleware-ethermind (maybe it's there but private).

**To Reproduce**
Use west to checkout the mcux-sdk repository e.g.

```
west init -m https://github.com/NXPmicro/mcux-sdk --mr MCUX_2.11.0 mcux-sdk
cd mcux-sdk
west update
```

**Expected behavior**
West checks out all the repositories without error.

**Screenshots and console output**
Finishes with the following output:

```
HEAD is now at 12e793e Add LA_OPT_NXP_Software_License and SCR file
ERROR: update failed for project mcux-sdk-middleware-ethermind
```

tgibson-work, 2022-07-29:
Semi-related - if I check out revision MCUX_2.11.0 this is not a revision of `main` branch, is there a 2.11.x revision that is on `main` branch, or does it not matter?

mcuxsusan, 2022-07-29:
Hi @tgibson-work, yes, that's a known issue because the mcux-sdk-middleware-ethermind repository has been moved to space https://github.com/NXP/ after the tag MCUX_2.11.0 is created. You could manually update west.yml to fix the issue.

```
--- a/west.yml
+++ b/west.yml
@@ -70,6 +70,7 @@ manifest:
     - name: mcux-sdk-middleware-ethermind
       path: middleware/wireless/ethermind
       revision: MCUX_2.11.0
+      remote: nxp
     - name: azure-rtos
       path: rtos/azure-rtos
       revision: MCUX_2.11.0
```

I am thinking to update the MCUX_2.11.0 tag to update west.yml for the change.

mcuxsusan, 2022-08-01:
Updated MCUX_2.11.0 tag and fixed the non-exist reference west.yml issue for mcux-sdk-middleware-ethermind.

closed 2022-08-01 by mcuxsusan

tgibson-work, 2022-08-01:
Thanks for fixing this!