nxp-mcuxpresso/legacy-mcux-sdk #91: Custom Board MCUX SDK support

raghulshanmuganathan-sigmaconnectivity, 2022-08-02:
Hi,

I have created a custom (out-of-tree) board for IMX RT 1061 based on MIMXRT1060_EVK Board. 
I followed all the steps from this guide https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html.

The application builds but does not link to the NXP HAL from modules directory and returns an error: "...  undefined reference to `dcd_data'" whenever I try to configure the CODE_LOCATION.

```
choice CODE_LOCATION
    default CODE_FLEXSPI
endchoice

```

I have also noticed that there is a CMakeLists and hal_nxp.cmake files in modules/hal/nxp/mcux folder which adds more configurations and includes directories based on the known NXP boards.

for example,
```
#include device specific drivers
if (${MCUX_DEVICE} MATCHES "MIMXRT1[0-9][0-9][0-9]")
  zephyr_include_directories(mcux-sdk/devices/${MCUX_DEVICE}/xip)
endif()
```

```
elseif (${MCUX_DEVICE} MATCHES "MIMXRT10[0-9][0-9]")
   include_driver_ifdef(CONFIG_PM_MCUX_GPC        gpc_1        driver_gpc_1)
   include_driver_ifdef(CONFIG_PM_MCUX_DCDC        dcdc_1        driver_dcdc_1)
   include_driver_ifdef(CONFIG_PM_MCUX_PMU        pmu        driver_pmu)
endif()

```

What is the easiest way to add a new custom (out-of-tree) board to these make files? 

I know I could probably create copies of the same files and use them in my board configuration. But it seems like a lot of work to make it work. I was wondering if NXP has a better, clean solution to do this.

mcuxsusan, 2022-08-03:
@dleach02, could you help the situation? Thanks in advance.

mmahadevan108, 2022-08-03:
Please submit a PR to the Zephyr HAL. Unfortunately there is no other easy way other than updating the files you mentioned.

dleach02, 2022-08-09:
This was handled in another thread: https://github.com/zephyrproject-rtos/zephyr/discussions/48746

closed 2022-08-09 by dleach02