nxp-mcuxpresso/legacy-mcux-sdk #94: [BUG]

tgibson-work, 2022-08-09:
**Describe the bug**
demo_apps fail to compile with `gcc-arm-none-eabi` 11.3.1 because of unknown type name `caddr_t` in `core/utilities/misc_utilities/fsl_sbrk.c`

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: MCUX_2.12.0
  - Toolchain: ARMGCC 11.3.Rel1 (11.3.1)
  - Board/SoC: EVKBIMXRT1050
  - Example hello_world
- Steps to reproduce the behavior:
  1. cd `/examples/evkbimxrt1050/demo_apps/hello_world/armgcc`
  2. Run `build_debug.sh` to build example, get compile error: 
```
   22 | caddr_t _sbrk(int incr);
      | ^~~~~~~
/home/user/git/nxp/mcux-sdk/core/utilities/misc_utilities/fsl_sbrk.c:23:1: error: unknown type name 'caddr_t'
```

**Expected behavior**
Build should compile without error

**Additional context**

Problem not present with ARMGCC 11.2.1

henric-jungheim, 2022-08-16:
We're also seeing this problem when trying to build a project based on the 1060 EVK sample source code when using the [11.3.Rel1 ARM GCC toolchain](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads).

mcuxsusan, 2022-08-17:
Thanks for the information of this issue, has reported to the development team and it's under analysis. Feedback may be delayed, appreciate for your patience.

mcuxsusan, 2022-08-19:
Update from the developer: caddr_t is defined in sys/types.h. For gcc 10, the stdio.h includes sys/types.h, in contrast, gcc 11 includes sys/_types.h. The fsl_sbrk.c include stdio.h, then it can't see caddr_t for gcc 11, need add explicitly.

Modification needed in fsl_sbrk.c:
```
@@ -8,6 +8,8 @@
 #if defined(__GNUC__)
 #include <stdio.h>
 #include <errno.h>
+#include <stdint.h>
+#include <sys/types.h>
 #endif
```

Fix will be available on github soon.

closed 2022-08-19 by mcuxsusan

mcuxsusan, 2022-08-22:
Fixed in https://github.com/NXPmicro/mcux-sdk/commit/197fbf8122ca960ad7517056ec938ea5871c13e5