nxp-mcuxpresso/legacy-mcux-sdk #95: SDK references incorrect MCU part for MIMXRT1060-EVKB

tgibson-work, 2022-08-11:
**Describe the bug**
The SDK XML for the MIMXRT1060-EVKB details the EVK as utilising the MIMXRT1062DVL6A package. This is incorrect as the EVK features the MIMXRT1062DVL6B package.

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: MCUX_2.12.0
  - Toolchain: N/A
  - Board/SoC: MIMXRT1060-EVKB
  - Example N/A
- Steps to reproduce the behavior:
  1. Import SDK to MCUXpresso
  2. Open SDK Import Wizard
  3. Find MIMXRT1060 part in list
  4. EVKB is incorrectly listed under MIMXRT1062xxxxA, it should be under MIMXRT1062xxxxB which has no entries. This is likely from the SDK XML which incorrectly lists `package="MIMXRT1062DVL6A"` under the `board` node for the MIMXRT1060-EVKB.

**Expected behavior**
EVKB should be listed under MIMXRT1062xxxxB entry in the SDK Sample Import Wizard. At present the MCU must be changed to the correct part after the sample has been imported, this should not be necessary.

**Additional context**

The "MIMXRT1060-EVKB Board User Manual" available at nxp.com correctly lists the board as featuring the MIMXRT1062DVL6B part, however the [product information page](https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/i-mx-rt1060-evaluation-kit:MIMXRT1060-EVKB) incorrectly lists the 'A' part as being fitted to the EVK.

mcuxsusan, 2022-08-11:
Thanks for the information, checking with the team and feedback may be delayed.

mcuxsusan, 2023-09-18:
The fix has been added in MCUX_2.14.0 release, latest main has integrated the change. Now selecting MIMXRT1062xxxxB only list board evkbmimxrt1060.

closed 2023-09-18 by mcuxsusan