nxp-mcuxpresso/legacy-mcux-sdk #106: [BUG] MCUX_2.13.0 is tagged wrongly

dnltz, 2023-02-08:
The tag for the v2.13.0 mcux release points to a commit that does not belong to main: https://github.com/nxp-mcuxpresso/mcux-sdk/tree/MCUX_2.13.0

I think it should point to 1df21e6e416f50e98bc4bb8e823adf2ce103f21d (https://github.com/nxp-mcuxpresso/mcux-sdk/commit/1df21e6e416f50e98bc4bb8e823adf2ce103f21d)

dnltz, 2023-02-08:
@mcuxcc FYI

mcuxcc, 2023-02-09:
Hi @dnltz 

I am sorry for the [1df216](https://github.com/nxp-mcuxpresso/mcux-sdk/commit/1df21e6e416f50e98bc4bb8e823adf2ce103f21d) commit log may confuse you.

Actually, it is a merge commit, merge 2.13.0 release to main branch.

As main branch is the develop branch,  in this branch user can request hotfix from NXP or contribute patch to this branch.
But the tag MCUX_2.13.0 is a commit point has been validated by NXP.

closed 2023-02-09 by dnltz