nxp-mcuxpresso/legacy-mcux-sdk #108: [BUG] SEMC_ConfigureDBI not setting REH2/REL2

marcoosi, 2023-03-06:
**Describe the bug**
According to `i.MX RT1060X Processor Reference Manual, Rev. 1, 05/2022`, the `DBICR1` register has 6 bits for REH and REL configuration. The `SEMC_ConfigureDBI` function in `fsl_semc.c` only writes the lower 4 bits.

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: master
  - Toolchain: any
  - Board/SoC: IMXRT1062
- Steps to reproduce the behavior:
  - Set `tRdxLow_Ns` in config struct to the value corresponding to 16 clock cycles of SEMC clock cycles will set the RDX Low Time to one clock cycle. 

**Expected behavior**
Given that the IMXRT106x have 6 bits to configure the RDX Low and High Times, these should be taken into account.

**Screenshots and console output**

**Additional context**

mcuxcc, 2023-03-08:
@marcoosi  thank you for reporting the issue.

i have forward the issue to internal team, i will update to you if any feedback

zejiang0jason, 2023-03-27:
Hi @marcoosi , what is the clkSrc_Hz, config->tRdxLow_Ns, and config->tRdxHigh_Ns in your case? thanks.

zejiang0jason, 2023-03-30:
Hi @marcoosi , please help to check the pull request for this issue, thanks:
https://github.com/nxp-mcuxpresso/mcux-sdk/pull/111

marcoosi, 2023-04-03:
@zejiang0jason the fix in #111 works for me.

PR #111 (2023-04-18): MCUXGH-646[SEMC] Fix SEMC_ConfigureDBI bug
**Prerequisites**

- [x] I have checked latest main branch and the issue still exists.
- [x] I did not see it is stated as known-issue in release notes.
- [x] No similar GitHub issue is related to this change.
- [x] My code follows the commit guidelines of this project.
- [x] I have performed a self-review of my own code.
- [x] My changes generate no new warnings.
- [x] I have added tests that prove my fix is effective or that my feature works.

**Describe the pull request**
Handle the DBICR1[REH2] and DBICR1[REL2] when the register bitfield exist.

F [... cut]
closed 2023-04-18 by mcuxcc