nxp-mcuxpresso/legacy-mcux-sdk #109: [BUG] Removing CMSIS is breaking our CMake build

CIPop, 2023-03-15:
**Describe the bug**
In the past release ([12](https://github.com/nxp-mcuxpresso/mcux-sdk/releases/tag/MCUX_2.12.0_UPDATE)), our sample at https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/NXP/mimxrt1060 was able to build as expected.

Trying to upgrade to the latest release ([13](https://github.com/nxp-mcuxpresso/mcux-sdk/releases/tag/MCUX_2.13.0_UPDATE)), our CMake generation fails to find the CMSIS targets for CortexM7 and DSP:

```
CMake Error at mimxrt1060/_deps/nxp_mcux_sdk-src/devices/MIMXRT1062/device_CMSIS.cmake:13 (include):
  include could not find requested file:

    CMSIS_Include_core_cm
Call Stack (most recent call first):
  mimxrt1060/_deps/nxp_mcux_sdk-src/drivers/common/driver_common.cmake:16 (include)
  mimxrt1060/_deps/nxp_mcux_sdk-src/drivers/igpio/driver_igpio.cmake:14 (include)
  demos/projects/NXP/mimxrt1060/CMakeLists.txt:53 (include)
[...]
CMake Error at demos/projects/NXP/mimxrt1060/CMakeLists.txt:85 (include):
  include could not find requested file:

    CMSIS_Include_core_cm
[...]
CMake Error at demos/projects/NXP/mimxrt1060/CMakeLists.txt:89 (include):
  include could not find requested file:

    CMSIS_Include_common
[...]
CMake Error at demos/projects/NXP/mimxrt1060/CMakeLists.txt:91 (include):
  include could not find requested file:

    CMSIS_Include_dsp
```

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: MCUX_2.13.0_UPDATE
  - Toolchain: Tested with 3 versions: ARMGCC 10.3.1; 11 and 12.
  - Board/SoC: MIMXRT1060-EVK
  - Example: CMSIS
- Steps to reproduce the behavior:
1. Change the `cmake/common/utilities.cmake` to use the latest MCUX-SDK within our repo: https://github.com/Azure-Samples/iot-middleware-freertos-samples
```
FetchContent_Declare(
    NXP_MCUX_SDK
    GIT_REPOSITORY https://github.com/NXPmicro/mcux-sdk.git
    GIT_TAG        3390e5498085ad152d318803a30924cd7199a56a
    GIT_PROGRESS   TRUE
)
```
2. Follow the steps here to build the repo for NXP: https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/NXP/mimxrt1060 

TL;DR with arm-gcc and Ninja installed:
```
git submodule update --init --recursive
cmake -G Ninja -DBOARD=mimxrt1060 -DVENDOR=NXP -Bmimxrt1060 .
cmake --build mimxrt1060
```

**Expected behavior**

Build succeeds.

**Screenshots and console output**
```
cmake -G Ninja -DBOARD=mimxrt1060 -DVENDOR=NXP -Bmimxrt1060 .
-- No NXP_TOOLCHAIN_PATH specified, using default: /usr
-- No NXP_TARGET_TRIPLET specified, using default: arm-none-eabi
-- No FREERTOS_PATH specified: Fetching FreeRTOS to path <>
-- Populating freertos
-- Configuring done
-- Generating done
-- Build files have been written to: S:/m/freertos/mimxrt1060/_deps/freertos-subbuild
[0/7] Performing update step for 'freertos-populate'
[1/7] No patch step for 'freertos-populate'
[3/7] No configure step for 'freertos-populate'
[4/7] No build step for 'freertos-populate'
[5/7] No install step for 'freertos-populate'
[6/7] No test step for 'freertos-populate'
[7/7] Completed 'freertos-populate'
INFOFREERTOS_PATH set to
-- Populating nxp_mcux_sdk
-- Configuring done
-- Generating done
-- Build files have been written to: S:/m/freertos/mimxrt1060/_deps/nxp_mcux_sdk-subbuild
[0/7] Performing update step for 'nxp_mcux_sdk-populate'
[1/7] No patch step for 'nxp_mcux_sdk-populate'
[3/7] No configure step for 'nxp_mcux_sdk-populate'
[4/7] No build step for 'nxp_mcux_sdk-populate'
[5/7] No install step for 'nxp_mcux_sdk-populate'
[6/7] No test step for 'nxp_mcux_sdk-populate'
[7/7] Completed 'nxp_mcux_sdk-populate'
INFONXP_MCUX_SDK_PATH set to
-- Populating lwip
-- Configuring done
-- Generating done
-- Build files have been written to: S:/m/freertos/mimxrt1060/_deps/lwip-subbuild
[0/7] Performing update step for 'lwip-populate'
[1/7] No patch step for 'lwip-populate'
[3/7] No configure step for 'lwip-populate'
[4/7] No build step for 'lwip-populate'
[5/7] No install step for 'lwip-populate'
[6/7] No test step for 'lwip-populate'
[7/7] Completed 'lwip-populate'
INFOLWIP_PATH set to
driver_phy-device-ksz8081 component is included.
driver_phy-common component is included.
driver_igpio component is included.
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMake Error at mimxrt1060/_deps/nxp_mcux_sdk-src/devices/MIMXRT1062/device_CMSIS.cmake:13 (include):
  include could not find requested file:

    CMSIS_Include_core_cm
Call Stack (most recent call first):
  mimxrt1060/_deps/nxp_mcux_sdk-src/drivers/common/driver_common.cmake:16 (include)
  mimxrt1060/_deps/nxp_mcux_sdk-src/drivers/igpio/driver_igpio.cmake:14 (include)
  demos/projects/NXP/mimxrt1060/CMakeLists.txt:53 (include)

driver_clock component is included.
utility_debug_console component is included.
component_serial_manager component is included.
component_lists component is included.
component_lpuart_adapter component is included.
driver_lpuart component is included.
driver_nor_flash-controller-flexspi component is included.
driver_flexspi component is included.
driver_nor_flash-common component is included.
component_serial_manager_uart component is included.
device_startup component is included.
device_system component is included.
driver_iomuxc component is included.
utility_assert component is included.
driver_xip_device component is included.
driver_xip_board component is included.
CMake Error at demos/projects/NXP/mimxrt1060/CMakeLists.txt:85 (include):
  include could not find requested file:

    CMSIS_Include_core_cm

driver_mdio-common component is included.
CMake Error at demos/projects/NXP/mimxrt1060/CMakeLists.txt:89 (include):
  include could not find requested file:

    CMSIS_Include_common

CMake Error at demos/projects/NXP/mimxrt1060/CMakeLists.txt:91 (include):
  include could not find requested file:

    CMSIS_Include_dsp

utilities_misc_utilities component is included.
driver_enet component is included.
driver_trng component is included.
driver_phy-common component is included.
driver_dcp component is included.
driver_cache_armv7_m7 component is included.
driver_snvs_hp component is included.
Vcpkg integrate step.
-- Read version from S:/m/freertos/libs/azure-iot-middleware-freertos/libraries/azure-sdk-for-c/sdk/inc/azure/core/az_version.h
-- AZ_SDK_VERSION_MAJOR 1
-- AZ_SDK_VERSION_MINOR 5
-- AZ_SDK_VERSION_PATCH 0
-- AZ_SDK_VERSION_PRERELEASE beta.1
-- Configuring incomplete, errors occurred!
See also "S:/m/freertos/mimxrt1060/CMakeFiles/CMakeOutput.log".
See also "S:/m/freertos/mimxrt1060/CMakeFiles/CMakeError.log".
```

CIPop, 2023-03-15:
> The project can work solely, if you only want to get the fundamental support for SoC(s) or board(s), you just use the original Git way to clone and checkout the project.

What are the steps to gain access to CMSIS cmake includes now that the `\CMSIS` folder was removed, without the usage of an external tool such as West?

mcuxcc, 2023-03-21:
Hi @CIPop 
Sorry for late reply
yes, we have changed cmsis from folder to a standalone repository in 2.13.0, you just need to 
1. update the mcux-sdk repository
2. run west update

Then the cmsis will be cloned into the same directory as before, could you please check if cmsis is cloned correctly?

Thanks

CIPop, 2023-03-21:
@mcuxcc I was able to run west and saw that it does what you described but in our repo we don't require/use west at all and we took your repo as a CMake dependency.
The recent changes are breaking our system and they seem to be in conflict with your statement "The project can work solely, ".

Please confirm that from 2.13 this repo isn't stand-alone and the use of `west` is mandatory.

mcuxcc, 2023-03-22:
Sorry for the incompatible change.
Compared to the folder structure before 2.13.0, user west is mandatory.

CIPop, 2023-03-22:
Thank you for confirming that `west` is now mandatory for your repo. 

Closing as by design.

closed 2023-03-22 by CIPop