nxp-mcuxpresso/legacy-mcux-sdk #110: [BUG] MIMXRT1011.h missing TCMSZ macros starting with 2.11.0

tannewt, 2023-03-21:
**Describe the bug**
The MIMXRT1011.h header should have the TCMSZ macros defined. They are used for dynamically setting the TCM size. Older versions before 2.11.0 have them.

**To Reproduce**

Use `IOMUXC_GPR_GPR14_CM7_CFGITCMSZ` to dynamically set the TCM size.

**Expected behavior**

The header should have these defines:

```

/*! @name GPR14 - GPR14 General Purpose Register */
/*! @{ */
#define IOMUXC_GPR_GPR14_CM7_CFGITCMSZ_MASK      (0xF0000U)
#define IOMUXC_GPR_GPR14_CM7_CFGITCMSZ_SHIFT     (16U)
/*! CM7_CFGITCMSZ
 *  0b0000..0 KB (No ITCM)
 *  0b0011..4 KB
 *  0b0100..8 KB
 *  0b0101..16 KB
 *  0b0110..32 KB
 *  0b0111..64 KB
 *  0b1000..128 KB
 */
#define IOMUXC_GPR_GPR14_CM7_CFGITCMSZ(x)        (((uint32_t)(((uint32_t)(x)) << IOMUXC_GPR_GPR14_CM7_CFGITCMSZ_SHIFT)) & IOMUXC_GPR_GPR14_CM7_CFGITCMSZ_MASK)
#define IOMUXC_GPR_GPR14_CM7_CFGDTCMSZ_MASK      (0xF00000U)
#define IOMUXC_GPR_GPR14_CM7_CFGDTCMSZ_SHIFT     (20U)
/*! CM7_CFGDTCMSZ
 *  0b0000..0 KB (No DTCM)
 *  0b0011..4 KB
 *  0b0100..8 KB
 *  0b0101..16 KB
 *  0b0110..32 KB
 *  0b0111..64 KB
 *  0b1000..128 KB
 */
#define IOMUXC_GPR_GPR14_CM7_CFGDTCMSZ(x)        (((uint32_t)(((uint32_t)(x)) << IOMUXC_GPR_GPR14_CM7_CFGDTCMSZ_SHIFT)) & IOMUXC_GPR_GPR14_CM7_CFGDTCMSZ_MASK)
/*! @} */
```

mcuxcc, 2023-03-22:
Thank you for issue reporting, i have forwarded the issue to internal team.

mcuxcc, 2023-03-28:
[image]
@tannewt  feedback from design team

mcuxsusan, 2023-05-22:
Close the issue because the soc header file is updated according to the latest RM update, users are not expected to touch this bits.

closed 2023-05-22 by mcuxsusan