nxp-mcuxpresso/legacy-mcux-sdk #113: LPSPI driver sometimes send bytes in WRONG ORDER

DRNadler, 2023-05-19:
Please let me know if this is a good place to report this problem.
I posted the details (including logic analyzer pictures and code to reproduce problem) on NXP support forum,
where again I got no response:
[i-MX-RT/LPSPI-driver-bug-Bytes-sent-IN-WRONG-ORDER](https://community.nxp.com/t5/i-MX-RT/LPSPI-driver-bug-Bytes-sent-IN-WRONG-ORDER/m-p/1645964#M24935)

- Environment (please complete the following information):
  - SDK_2_13_0_MIMXRT1024xxxxx
  - Toolchain: MCUXpresso IDE v11.7.0 [Build 9198] [2023-01-17]

Thanks!
Best Regards, Dave

mcuxsusan, 2023-05-23:
Thanks for reporting the issue, already forwarded the issue to internal team, feedback could be delayed.

mcuxthomas, 2023-07-13:
Hi @DRNadler :
[i-MX-RT/LPSPI-driver-bug-Bytes-sent-IN-WRONG-ORDER](https://community.nxp.com/t5/i-MX-RT/LPSPI-driver-bug-Bytes-sent-IN-WRONG-ORDER/m-p/1645964#M24935)

[image]

Hi We have a repo in github https://github.com/nxp-mcuxpresso/mcux-sdk  

You can create a PR , Thanks .

[@davenadler](https://community.nxp.com/t5/user/viewprofilepage/user-id/132425) 

https://github.com/nxp-mcuxpresso/mcux-sdk/blob/main/drivers/lpspi/fsl_lpspi.c

DRNadler, 2023-07-14:
6 weeks after I post an issue to this repository, you reply to tell me there's a repository here? Really now...

Further testing shows erroneous byte-swapping also happens on receive side.
Does anybody ever test this stuff?

mcuxthomas, 2023-07-17:
> 6 weeks after I post an issue **to this repository**, you reply to tell me there's a repository here? Really now...
> 
> Further testing shows erroneous byte-swapping also happens on receive side. Does anybody ever test this stuff?

Could you tell  how to fix this function?
```
static uint32_t LPSPI_CombineWriteData(uint8_t *txData, uint8_t bytesEachWrite, bool isByteSwap)
{
    assert(txData != NULL);

    uint32_t wordToSend = 0U;

    switch (bytesEachWrite)
    {
        case 1:
            wordToSend = *txData;
            ++txData;
            break;

        case 2:
            if (!isByteSwap)
            {
                wordToSend = *txData;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 8U;
                ++txData;
            }
            else
            {
                wordToSend = (unsigned)(*txData) << 8U;
                ++txData;
                wordToSend |= *txData;
                ++txData;
            }

            break;

        case 3:
            if (!isByteSwap)
            {
                wordToSend = *txData;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 8U;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 16U;
                ++txData;
            }
            else
            {
                wordToSend = (unsigned)(*txData) << 16U;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 8U;
                ++txData;
                wordToSend |= *txData;
                ++txData;
            }
            break;

        case 4:
            if (!isByteSwap)
            {
                wordToSend = *txData;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 8U;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 16U;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 24U;
                ++txData;
            }
            else
            {
                wordToSend = (unsigned)(*txData) << 24U;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 16U;
                ++txData;
                wordToSend |= (unsigned)(*txData) << 8U;
                ++txData;
                wordToSend |= *txData;
                ++txData;
            }
            break;

        default:
            assert(false);
            break;
    }
    return wordToSend;
}

static void LPSPI_SeparateReadData(uint8_t *rxData, uint32_t readData, uint8_t bytesEachRead, bool isByteSwap)
{
    assert(rxData != NULL);

    switch (bytesEachRead)
    {
        case 1:
            *rxData = (uint8_t)readData;
            ++rxData;
            break;

        case 2:
            if (!isByteSwap)
            {
                *rxData = (uint8_t)readData;
                ++rxData;
                *rxData = (uint8_t)(readData >> 8);
                ++rxData;
            }
            else
            {
                *rxData = (uint8_t)(readData >> 8);
                ++rxData;
                *rxData = (uint8_t)readData;
                ++rxData;
            }
            break;

        case 3:
            if (!isByteSwap)
            {
                *rxData = (uint8_t)readData;
                ++rxData;
                *rxData = (uint8_t)(readData >> 8);
                ++rxData;
                *rxData = (uint8_t)(readData >> 16);
                ++rxData;
            }
            else
            {
                *rxData = (uint8_t)(readData >> 16);
                ++rxData;
                *rxData = (uint8_t)(readData >> 8);
                ++rxData;
                *rxData = (uint8_t)readData;
                ++rxData;
            }
            break;

        case 4:
            if (!isByteSwap)
            {
                *rxData = (uint8_t)readData;
                ++rxData;
                *rxData = (uint8_t)(readData >> 8);
                ++rxData;
                *rxData = (uint8_t)(readData >> 16);
                ++rxData;
                *rxData = (uint8_t)(readData >> 24);
                ++rxData;
            }
            else
            {
                *rxData = (uint8_t)(readData >> 24);
                ++rxData;
                *rxData = (uint8_t)(readData >> 16);
                ++rxData;
                *rxData = (uint8_t)(readData >> 8);
                ++rxData;
                *rxData = (uint8_t)readData;
                ++rxData;
            }
            break;

        default:
            assert(false);
            break;
    }
}
```

DRNadler, 2023-07-17:
On 7/16/2023 10:32 PM, @mcuxthomas  Thomas Li wrote:
**_> Could you tell how to fix this function?_**

In the functions **LPSPI_CombineWriteData**  and **LPSPI_SeparateReadData** ,
change every **!isByteSwap** to **isByteSwap** .
Then find and fix the other bug: when continuous mode is selected,
the behavior is different, which is a bug...

mcuxthomas, 2023-07-18:
Hi @DRNadler :
The byte swap is because bitsPerFrame=16 in the second test, not related to the continuous mode. Configuration flag kLPSPI_MasterByteSwap can be used to control the byte order, please see the comment:
https://github.com/nxp-mcuxpresso/mcux-sdk/blob/5e915fe4cfc1a30b0f455286027e29d3f805cf60/drivers/lpspi/fsl_lpspi.h#L211-L223

In your case, kLPSPI_MasterByteSwap need be set, please try tiConfigFlag = kLPSPI_MasterPcs1 | kLPSPI_MasterByteSwap

like this:
[image]

[image]

mcuxsusan, 2023-09-18:
@DRNadler, any feedback on above suggestion? Did you have your project working as expected?

DRNadler, 2023-10-17:
Folks, please try explain this:

Requirements, for **_many_** different SPI devices in our current project, and **_many_** of your customers:
1) Bytes must be read from and written to SPI **in the same byte-order as in memory** (**_no_** byte reversal),
2) CS must be continuously asserted over the multi-byte transfer
3) the number of bytes varies and may or may not be a multiple of 2 or 4.

How does on do this with your driver???
Thanks,
Best Regards, Dave

PS: Example above both reads and sends bytes in wrong order (unless kLPSPI_MasterByteSwap is set, which is completely backwards from what users expect).

mcuxthomas, 2023-10-23:
Dear @DRNadler :
Thank you.
This is the reslolution of our driver.
1. Bytes must be read from and written to SPI in the same byte-order as in memory (no byte reversal),
  => Just use the kLPSPI_MasterByteSwap flag . Here is the example ref: 

https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/1db053064902221473bdbda64fa4bd81a07f5dca/evkbmimxrt1170/driver_examples/lpspi/polling_b2b_transfer/master/cm7/lpspi_polling_b2b_transfer_master.c#L113-L114

You can get this example in every platform SDK.
2. CS must be continuously asserted over the multi-byte transfer
=> CS can be continuously asserted by flag kLPSPI_MasterPcsContinuous , same as example as follow:
examples/blob/1db053064902221473bdbda64fa4bd81a07f5dca/evkbmimxrt1170/driver_examples/lpspi/polling_b2b_transfer/master/cm7/lpspi_polling_b2b_transfer_master.c#L113-L114

3. the number of bytes varies and may or may not be a multiple of 2 or 4.
 => the number of bytes can be varies by bitsPerFrame . The default value is 8bit , just 1BYTE. You can set some other bytes. 
https://github.com/nxp-mcuxpresso/mcux-sdk/blob/6e0302f248da600dfa07ab5f780e7857ce54bca9/drivers/lpspi/fsl_lpspi.c#L353C19-L353C32

If you face some device can't be set varies number you can give the information of the device . We can try to repeat it .

BTW, A lot of our customers are already using kLPSPI_MasterByteSwap. Just set kLPSPI_MasterByteSwap the order will be fine.

We appreciate your feedback. Thank you very much.

DRNadler, 2023-11-27:
@mcuxthomas - Firstly, you are confusing "CS must be asserted across an SPI transfer operation" and "CS left asserted between SPI transfer operations". This is aggravated by poor choice of naming and terminology in the API and datasheet.

More importantly, to summarize what you have written above:

**_In order to prevent byte order swapping, you must set the kLPSPI_MasterByteSwap flag._**

How do you not understand that this is backwards and a bug?
Really now...

zejiang0jason, 2023-11-28:
Hi @DRNadler , 

> Firstly, you are confusing "CS must be asserted across an SPI transfer operation" and "CS left asserted between SPI transfer operations". This is aggravated by poor choice of naming and terminology in the API and datasheet.

Which document results to the confusion? We will update it.

About the swap, my understanding is, the code can generate the desired waveform, your concern is the flag backwards.  If so, I suggest keep current implementation, so that user code which have been adapted to this driver won't be impacted. The flag `kLPSPI_MasterByteSwap`'s comment has explained the behavior, user can select based on the case requirement.

Thanks.