nxp-mcuxpresso/legacy-mcux-sdk #118: [BUG] QN9090DK6 Board is Missing

ciband, 2023-06-02:
**Describe the bug**

The SDK for the QN9090DK6 is not in this repository.  This makes the VSCode extension not usable for this platform.  Could this SDK and associated support be added?

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: main branch
  - Toolchain: N/A
  - Board/SoC: QN9090DK6
  - Example N/A
- Steps to reproduce the behavior:

1. Import this Github SDK into the VSCode extension for MCUXpresso.
2. Observe that the QN9090 SDK nor the QN9090DK6 board is available for selection.

**Expected behavior**

The QN9090DK6 board (and the processor in general) should be supported.

**Screenshots and console output**

N/A

**Additional context**

If I have somehow missed this support and this is PEBKAC, my apologies, and I would appreciate any direction to get this working.  Thank you.

mcuxsusan, 2023-06-05:
Hi @ciband, this seems to be a new feature but not bug. I will check with the team to see if there's plan to release the QN9090DK6 board in GitHub SDK.

mcuxsusan, 2023-07-05:
The QN9090 board is now added, please checkout to https://github.com/nxp-mcuxpresso/mcux-sdk/commit/72ba5331d5f7b2ae465b55774db9e65207f4e5d4 and run "west update" to get the SDK for this board.

closed 2023-07-05 by mcuxsusan