nxp-mcuxpresso/legacy-mcux-sdk #119: CMSIS macro usage

david-hazi-arm, 2023-06-14:
Dear NXP,

Would it be possible to use CMSIS macros instead of hardcoding GCC like variables (for example __Vectors).
If we want to use the SDK file in another project, it could cause some issues if we want to compile the SDK's source files with IAR (Armclang only works, because it using the same naming), like you can see it in TF-M:
https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/tree/platform/ext/target/nxp/lpcxpresso55s69/Device/Source/startup_lpcxpresso55s69.c#n226

Thanks,
Dávid Házi

mcuxsusan, 2023-06-16:
Hi @david-hazi-arm, thanks! Issue has been forwarded to development team and feedback could be delayed, appreciate for your patience.

mcuxsusan, 2023-06-20:
Hi @david-hazi-arm, checked with the team, NXP currently provide toolchain specific startup files so maybe toolchain macros/variables are used, that's why you have issue porting the GCC startup file to IAR toolchain. You have made a good suggestion in this issue inspiring the team to consider whether to provide a unified startup c file across different toolchains, thanks. 

Backing to this issue itself, in short term we may not able to deliver a fix in this repository, we appreciate if you would like to raise a pr to contribute.