nxp-mcuxpresso/legacy-mcux-sdk #120: [BUG] i.MX8ML generated project doesn't build under MCUXpresso IDE

bizulk, 2023-06-14:
**Describe the bug**

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: main 58ae1de318b
  - Toolchain: gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10)
  - Board/SoC: variscite i.MX8Mplus SOM : VAR-SOM-IMX8Mplus with symphony board
  - Example : new C or C++ project template
- Steps to reproduce the behavior:
  1. from installed SDK select MIXM8MLxxxLZ device and right-click --> select create project.
  2. create C or C++ project.
  3. optionally select newlib instead redlib, select UART driver, rpmsg middleware.
  4. launch the build.

**Expected behavior**
Build should complete and produce the axf executable.

**Screenshots and console output**

In summary :: 
many files not found because a missing copy from the sdk or their names are incorrect.
  1. MIMX8ML8.h not found : shoudl be MIMX8LM8_cm7.h
  2. rpmsg_config not found : because a define RS_USE_CUSTOM_CONFIG is set but there is not custom config available. Default config should be set and this symbol not set in the file.
  3. fsl_iomux.c  not found : this file comes from the sdk (core/drivers/mu). the .h/.c pair should be copied.
  4. fsl_mu.h not found : same a fsl_iomux.
  5. warning cannot find entry symbol ResetIRS :the startup file is empty. I checked the SDK for a startup file they are all the same for the i.MX8Mblabla. So the file should be copied from there.
  6. I am not sure about the linker file. It is very different from the SDK examples, those allow me to build binaries for DDR/TCRAM/Flash. It seems that the project targets the TCRAM.

After that the build succeeded but I can't get the Jlink debugger working. Is the NXP  [patch](https://www.nxp.com/webapp/Download?colCode=SDK_MX8MP_3RDPARTY_Patch&appType=license) integrated in the launch configuration ?
I am not very familiar with this debugger. I checked that the firmware is up to date and its' configuration : 
[image]

I checked in SEGGER [wiki](https://www.segger.com/supported-devices/nxp) that this device is supported.

Also I would like a clarification about this repository : it is an official NXP repository ? 

Best regards.

**Additional context**

mcuxsusan, 2023-06-16:
Hi @bizulk, yes, this is NXP official repository for demonstrating the MCUXpresso SDK. I have forwarded the issue to internal team and feedback will be provided later, appreciate for your patience.

bizulk, 2023-06-19:
Hi,
thanks so much at least for the confirmation.
Also I would like to report that I managed to make the debugging working by make a change in the device selection.

The console was reporting : 

```
Command line: -SettingsFile "/home/nautilus/dev/phrt-workspace/MIMX8ML8_Project/Debug/MIMX8ML8_Project JLink Debug 
SettingsFile.jlink" -nosilent -swoport 2332 -select USB=203200510 -telnetport 2333 -singlerun -endian little -noir -speed auto -port 2331 -vd -device MIMX8ML8xxxLZ -if JTAG -halt -reportuseraction
-----GDB Server start settings-----
Failed to get index for device name 'MIMX8ML8xxxLZ'.GDBServer will be closed...
```

Then I made a change on the device selection as follow, basing on variscite demo's using vscode . 

[image]

mcuxsusan, 2023-07-05:
Hi @bizulk, thanks for reporting the issue and continuous follow up. The root cause for series of the issue is that the MIMX8ML8 software release does not support MCUX IDE. Then in software validation cycle the MCUX IDE new project creation feature is not validated, thus you meet so many issue for building new project and debugging this new project. Apologize for the bad experience here. In future IDE may limit the feature when detect a device does not support MCUX IDE.

closed 2023-07-05 by mcuxsusan

bizulk, 2023-07-05:
Hello.
I'm confused, may I ask a question about the development workflow for the CM7 in the MIMX8ML8 ?
How do I configure the pinmux, the controller, the clocks, the std library, the middleware to embbed ? 
The getting started for this device is pretty much limited to build and launch example application.

mcuxsusan, 2023-07-10:
Creating a new project from scratch could be a little bit difficult, I would like to recommend you to start on an existing example project. Using NXP provided [MCUXpresso-Config-Tools](https://www.nxp.com/design/software/development-software/mcuxpresso-software-and-tools-/mcuxpresso-config-tools-pins-clocks-and-peripherals:MCUXpresso-Config-Tools), you will be able to clone an existing project in SDK package for you to start, you will also be able to import the existing pin_mux.c/h file into the config tool to re-configure the pinmux, import the existing clocks to re-configure the clock.