nxp-mcuxpresso/legacy-mcux-sdk #121: Caam output buffer not aligned

GiulioDallaVecchia, 2023-06-19:
**Describe the bug**

With MBEDTLS (2.28.1) library when you try to handshake a connection is computed a SHA256. Ouput buffer of the caam peripheral is not aligned to 32bit.

When function names "ssl_prepare_server_key_exchange" is called "unsigned char hash[MBEDTLS_MD_MAX_SIZE]" is allocated on the stack, but is not aligned. (file ssl_srv.c, row number 3519.)

mcuxsusan, 2023-06-20:
Hi @VitChmelarNXP, could you please help check if this issue relates to mbedtls and should be raised to mbedtls repository? Thanks a lot.

PatrikNemethNXP, 2023-08-14:
Hi @GiulioDallaVecchia, I could not reproduce this issue on my end. Could you please provide additional information on your environment, such as the used board, compiler / IDE and any specific compiler settings?

PatrikNemethNXP, 2024-01-23:
I will close the issue for now as not reproducible. The issue can be reopened if it persists and more information is gathered.

closed 2024-01-23 by PatrikNemethNXP