nxp-mcuxpresso/legacy-mcux-sdk #122: Build SDK on top of CMSIS-RTOS2 interface

escherstair, 2023-06-29:
mcux-sdk is heavily coupled to FreeRTOS as the only possible RTOS.
But the choice of an RTOS depends on several conditions and needs.
FreeRTOS is quite popular, but other choices are Zephyr, RT-Thread, QNX, ThreadX/Azure RTOS, ...

Working on Cortex processor there is a great interface for RTOS handling and it's the [CMSIS-RTOS2](https://arm-software.github.io/CMSIS_5/RTOS2/html/index.html).

> It provides a standardized API for software components that require RTOS functionality and gives therefore serious benefits to the users and the software industry

If mcux-sdk woiuld be based on CMSIS-RTOS2 (with FreeRTOS under the hood), every user would be free to change to another RTOS easily.
The benefits for the user are great!

Would NXP think it would be possible?

mcuxsusan, 2023-11-29:
Hi @escherstair, thanks for the suggestion. Sorry for the late reply, I have forwarded the suggestion to internal team and did not get reply yet. Will come to you once I got feedback.

DavidLit, 2024-06-15:
Any progress on this?  CMSIS-RTOS2 is *so* much cleaner than the grossly overwrought FreeRTOS API!

ST's had RTOS2 for *years*!  Where are you guys?

Badokas, 2024-06-15:
+1
I agree, any news on this? `mcux-sdk` feels outdated in this regard.