nxp-mcuxpresso/legacy-mcux-sdk #129: west.yml: URL for rpmsg_lite is redundant as nxpmicro is anyway default remote

s-hemer, 2023-07-17:
This is just cosmetics but it would be only deleting this line: https://github.com/nxp-mcuxpresso/mcux-sdk/blob/5e915fe4cfc1a30b0f455286027e29d3f805cf60/west.yml#L38

s-hemer, 2023-07-17:
Ok, now I myself got confused. Deleting worked as https://github.com/NXPmicro/rpmsg-lite exists, yet the potentially correct solution should be replacing the line with `remote: nxp-mcuxpresso` obviously.

mcuxsusan, 2023-07-21:
Thanks for suggesting this. Per my understanding, west will take URL assignment as priority when a project has URL key. But, yes you are right the rgmsg-lite repository is also located in remote: nxp-mcuxpresso, then remove the URL line also works because when URL is not defined, the project will fall to use the default remote nxp-mcuxpresso.

s-hemer, 2023-07-28:
Of course it works anyways. As I wrote, it is a matter of cosmetics/cleanliness. If one has defined a remote, it is confusing to read the same one as URL again and you just wonder if you missed something.

commit 3f6a75a5 (2023-09-18): fixes #129 west: remove redundant url for rpmsg_lite
closed 2023-09-18 by McuxCIBot