nxp-mcuxpresso/legacy-mcux-sdk #131: [BUG] CLOCK_EnableUsbhs0/1PhyPllClock uses PWD_SET instead of PWD register

c-mauderer, 2023-08-04:
The CLOCK_EnableUsbhs0PhyPllClock and CLOCK_EnableUsbhs1PhyPllClock functions try to set the PWD of the USBPHY to 0 with lines like this:

    USBPHY1->PWD_SET  = 0x0;

This won't have an effect because the `PWD_SET` register is used. Instead either `PWD` should be set to `0` or `PWD_CLR` should be set to `0xffffffff`.

The lines are here:

https://github.com/nxp-mcuxpresso/mcux-sdk/blob/ba0921b447a66e6ef1bb59c9fa2c5494a0915597/devices/MIMXRT1166/drivers/fsl_clock.c#L1738

https://github.com/nxp-mcuxpresso/mcux-sdk/blob/ba0921b447a66e6ef1bb59c9fa2c5494a0915597/devices/MIMXRT1166/drivers/fsl_clock.c#L1844

mcuxsusan, 2023-09-04:
Thanks for reporting the issue, already asked development team to check, reply could be delayed.

mcuxsusan, 2023-09-21:
The issue has been fixed in latest main branch, which integrate the MCUX_2.14.0 release update.
I see the PWD register operation is removed. https://github.com/nxp-mcuxpresso/mcux-sdk/blob/3f6a75a587dc64f07e852dbf484291ff9a6c3592/devices/MIMXRT1166/drivers/fsl_clock.c#L1660

closed 2023-09-21 by mcuxsusan