nxp-mcuxpresso/legacy-mcux-sdk #136: There is no manifest for FRDM-K64F and therefore no way to import FRDM-K64F examples.

iustmehri, 2023-09-03:
**Is your enhancement proposal related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

mcuxsusan, 2023-09-04:
Hi @iustmehri, could you please help provide more details? Which IDE are you using to explore our sdk? The MCUXpresso IDE or our mcuxpresso vscode extension? I assumed that you have followed the suggestion to use west tool to get our full delivery of SDK. In the west way of getting full sdk, the manifest file for FRDM-K64F could be found in the **examples/manifests** path. If you are using MCUXpresso IDE, could you please select manifest path to examples/manifest when importing sdk as git repository? See below pic as a reference:
[image]

iustmehri, 2023-09-13:
Hello @mcuxsusan! Thanks for the answere. I use the mcuxpresso vscode extension and to get the SDK I use the interface provided by the extension.

iustmehri, 2023-09-21:
Thank you so much @mcuxsusan

xtrollyj00, 2023-10-17:
Still experiencing the same problem, I am using extension for VSCode.

mcuxsusan, 2023-10-18:
Hi @xtrollyj00, checking with the team on the release plan of VSCode extension update which integrates the manifests selection. I will share with you the plan once I get the information.

mcuxsusan, 2023-10-19:
> Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

Checked with the team, the next release with the update integrated targets the end of October.

Thanks,
Susan

xtrollyj00, 2023-10-22:
Thank you @mcuxsusan.

mcuxsusan, 2023-11-13:
I checked the latest vscode extension(v1.3.59) and find the FRDM-K64F board is now displayed, see as below:
[image]

closed 2023-11-13 by mcuxsusan