nxp-mcuxpresso/legacy-mcux-sdk #143: [BUG] Some FreeRTOS files required for examples are not imported correctly

stefanct, 2023-09-29:
When importing FreeRTOS-based examples from the SDK the build fails because some of the FreeRTOS core files are not copied or referenced in the created project. I couldn't [get the SDK to install/work at all from ZIPs](https://community.nxp.com/t5/i-MX-Processors/SDK-failure/m-p/1729052#M213123) so this is all based on a `west` checkout.

Since I have not investigated in detail how the examples are imported into mcuxpresso I am not entirely sure if this is a bug in the IDE, the SDK, or the examples.

- Environment (please complete the following information):
  - MCUXpresso IDE v11.8.0
  - Tag/Commit hash: current HEAD (1d72f41b03f0d7b06920f3d5211391b030fa2dd8)
  - Toolchain: mcuxpresso's gcc
  - Board/SoC: initially MIMXRT1024-EVK but it does not seem to be hardware-dependent. frdmk32l3a6 and lpcxpresso54s018m behave exactly the same.
  - Example any based on FreeRTOS, I'll use `dev_dfu_freertos` as an example below.
- Steps to reproduce the behavior:
  1. From the quickstart menu select "Import SDK example(s)..."
  2. Choose any hardware, e.g., `evkmimxrt1024` 
  3. Choose and FreeRTOS-based example, e.g., `dev_dfu_freertos` and finish the import
  4. Make sure the right project is selected and try to build it.

**Expected behavior**
The build finishes including the linker step.

**Screenshots and console output**
However, linking fails because of missing functions:

    Invoking: MCU Linker
    arm-none-eabi-gcc -nostdlib -Xlinker -no-warn-rwx-segments -Xlinker -Map="ex-_dev_dfu_freertos.map" -Xlinker --gc-sections -Xlinker -print-memory-usage -Xlinker --sort-section=alignment -Xlinker --cref -mcpu=cortex-m7 -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -T ex_dev_dfu_freertos_Debug.ld -o "ex-_dev_dfu_freertos.axf" ./xip/evkmimxrt1024_flexspi_nor_config.o ./xip/fsl_flexspi_nor_boot.o  ./utility/usb_dfu_ram.o  ./utilities/fsl_assert.o ./utilities/fsl_debug_console.o ./utilities/fsl_memcpy.o ./utilities/fsl_str.o  ./usb/phy/usb_phy.o  ./usb/device/source/ehci/usb_device_ehci.o  ./usb/device/source/usb_device_ch9.o ./usb/device/source/usb_device_dci.o  ./usb/device/class/dfu/usb_device_dfu.o  ./usb/device/class/usb_device_class.o  ./startup/startup_mimxrt1024.o  ./source/dfu.o ./source/dfu_app.o ./source/dfu_timer.o ./source/semihost_hardfault.o ./source/usb_device_descriptor.o  ./freertos/freertos-kernel/portable/MemMang/heap_4.o  ./freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.o  ./drivers/fsl_cache.o ./drivers/fsl_clock.o ./drivers/fsl_common.o ./drivers/fsl_common_arm.o ./drivers/fsl_gpio.o ./drivers/fsl_lpuart.o ./drivers/fsl_pit.o  ./device/system_MIMXRT1024.o  ./component/uart/fsl_adapter_lpuart.o  ./component/timer/fsl_adapter_pit.o  ./component/serial_manager/fsl_component_serial_manager.o ./component/serial_manager/fsl_component_serial_port_uart.o  ./component/osa/fsl_os_abstraction_free_rtos.o  ./component/lists/fsl_component_generic_list.o  ./board/board.o ./board/clock_config.o ./board/dcd.o ./board/pin_mux.o   
    /usr/local/mcuxpressoide-11.8.0_1165/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.8.0.202306131047/tools/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: ./freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.o: in function `SVC_Handler':
    .../Debug/../freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.c:252: undefined reference to `pxCurrentTCB'
    /usr/local/mcuxpressoide-11.8.0_1165/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.8.0.202306131047/tools/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: ./freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.o: in function `PendSV_Handler':
    .../Debug/../freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.c:446: undefined reference to `vTaskSwitchContext'
    /usr/local/mcuxpressoide-11.8.0_1165/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.8.0.202306131047/tools/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: .../Debug/../freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.c:446: undefined reference to `pxCurrentTCB'
    /usr/local/mcuxpressoide-11.8.0_1165/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.8.0.202306131047/tools/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: ./freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.o: in function `SysTick_Handler':
    .../Debug/../freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.c:508: undefined reference to `xTaskIncrementTick'
    /usr/local/mcuxpressoide-11.8.0_1165/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.8.0.202306131047/tools/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: ./source/dfu_app.o: in function `main':
    .../Debug/../source/dfu_app.c:360: undefined reference to `xTaskCreate'
    /usr/local/mcuxpressoide-11.8.0_1165/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.8.0.202306131047/tools/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: .../Debug/../source/dfu_app.c:376: undefined reference to `vTaskStartScheduler'
    /usr/local/mcuxpressoide-11.8.0_1165/ide/plugins/com.nxp.mcuxpresso.tools.linux_11.8.0.202306131047/tools/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: ./component/osa/fsl_os_abstraction_free_rtos.o: in function `OSA_MutexCreate':
    .../Debug/../component/osa/fsl_os_abstraction_free_rtos.c:546: undefined reference to `xQueueCreateMutex'

**Additional context**
The reason for this is that the core FreeRTOS files are missing from the project, i.e., those mentioned in https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/main/manifests/MIMXRT1024-EVK_manifest_v3_13.xml:

    34077-      <source relative_path="./" type="src">
    34078-        <files mask="croutine.c"/>
    34079-        <files mask="event_groups.c"/>
    34080-        <files mask="list.c"/>
    34081:        <files mask="queue.c"/>
    34082-        <files mask="stream_buffer.c"/>
    34083-        <files mask="tasks.c"/>
    34084-        <files mask="timers.c"/>
    34085-      </source>

If I copy them manually into the project then it builds fine(!)

mcuxsusan, 2023-10-19:
> Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

Thanks for reporting the issue. I have reproduced the issue using the MCUXpresso IDE with manifests folder setting to core/manifests, will investigate and provide update once I get a fix.
I also found this issue does not exist when using manifests folder examples/manifests, could you please help give a try?
***@***.***

Thanks,
Susan

mcuxsusan, 2023-11-28:
Hi @stefanct, the issue import/build/clone with MCUX IDE using the core/manifests has been fixed in latest main branch at 80e20cde0f23217c64f043362f142a9bc010abd3. If you are working on the main branch, you could pull the core to latest and then run 'west update' to check the update.
```
cd mcuxsdk/core
git fetch
git rebase
west update
```

closed 2023-11-28 by mcuxsusan