nxp-mcuxpresso/legacy-mcux-sdk #145: [BUG] log and log_freertos hardfault immediately after boot

orlyprofili, 2023-10-19:
**Describe the bug**
log and log_freertos hardfault immediately after boot

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: SDK 2.14.0
  - Toolchain: GCC w/SDK_2_14_0_MIMXRT1170-EVKB
  - Environment: Macos 13.15.1, vscode mcuxpresso plugin
  - Board/SoC: MIMXRT1170-EVKB w/ DAP debugger
  - Example:  log and log_freertos

- Steps to reproduce the behavior:
    - Build
    - Debug
    - Hardfault
    - Other examples are fine

mcuxsusan, 2023-10-19:
> Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

Thanks for reporting the issue. I have already forwarded the issue to the development team and the reply could be delayed.

Thanks,
Susan

mcuxsusan, 2023-11-29:
Not sure if this relates to the macOS and vscode plugin issue, because the team cannot reproduce the issue running with MCUXIDE under windowsOS. I will involve tool team to help check.