nxp-mcuxpresso/legacy-mcux-sdk #150: [BUG] MCG_C4[DMX32] and MCG_C4[DRST_DRS] are not updated in CLOCK_SetFbeMode()

VoodooChild99, 2023-11-17:
**Describe the bug**
In the current implementation of `CLOCK_SetFbeMode()`, `MCG_C4[DMX32]` and `MCG_C4[DRST_DRS]` are not updated to the given value. For example, in the following code snippet, the local variable `mcg_c4` is updated, but it is not written into `MCG->C4`:
https://github.com/nxp-mcuxpresso/mcux-sdk/blob/main/devices/MK64F12/drivers/fsl_clock.c#L1810C5-L1817C6

Initially I thought this might be a workaround to ERR007993. However, after reading about ERR007993, I found the following sentence: "If you want to change the MCG_C4[DMX32] or MCG_C4[DRST_DRS] to new values along with the reference clock, the sequence described above must be performed before setting these values to the new value(s)". The suggested sequence has been performed, but C4 is not updated, therefore I believe this is a bug.

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: cc52d8e788059f83b1de680013aac5ff0f08d4c4
  - Board/SoC: FRDM-K64F, FRDM-K22F, and many more
  - Example: https://github.com/nxp-mcuxpresso/mcux-sdk/blob/main/devices/MK64F12/drivers/fsl_clock.c#L1810C5-L1817C6

**Expected behavior**
`MCG_C4[DMX32]` or `MCG_C4[DRST_DRS]` should be updated after `CLOCK_SetFbeMode()`

**Screenshots and console output**

**Additional context**

mcuxsusan, 2023-11-17:
Thanks for reporting the issue. The issue has been forwarded to development team for investigation, reply could be delayed, thanks for your patience.

Thanks,
Susan

mcuxsusan, 2023-12-05:
Update: The developer has confirmed it's an issue and working on it.

commit 43a5a8df (2024-11-28): fixes #150: MCG_C4[DMX32] and MCG_C4[DRST_DRS] are not updated

- Set C4 register after related variablle is updated;
- Updated copyright to 2023;
- Updated driver version.
closed 2023-12-06 by McuxCIBot