nxp-mcuxpresso/legacy-mcux-sdk #158: Include up-to-data CMSIS-DSP in SDK

escherstair, 2024-01-09:
SDK 2.14.0 uses the [https://github.com/ARM-software/CMSIS_5/releases/tag/5.9.0](https://github.com/ARM-software/CMSIS_5/releases/tag/5.9.0) but a couple of yesr ago CMSIS-DSP has been moved off the CMSIS base pack as below information
[https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP)

This change has not been considered yet in the SDK and so even the newest SDK 2.14.0 included old CMSI-DSP 1.9.0 (the last release before CMSIS-DSP moving).
At the moment [CMSIS-DSP](https://github.com/ARM-software/CMSIS-DSP) is at release 1.15.0

SDK should point to [CMSIS-DSP](https://github.com/ARM-software/CMSIS-DSP) repo and so it should include the latest release.
Based on a private support ticket, this is planned for next major update, scheduled for 2024 H2.
I opened this ticket to monitor this task in a public way.

mcuxsusan, 2024-01-09:
Thank you for reporting the issue. Will check the issue with develop team, feedback could be delayed.

escherstair, 2024-07-23:
@mcuxsusan
An update on this issue:
The latest SDK 2.16.0 updated CMSIS-DSP to 1.10.0 and not to the up-to-date [1.15.0](https://github.com/ARM-software/CMSIS-DSP/releases/tag/v1.15.0) (1.16.0 has been released some days ago, and so it's clear that it cannot be included yet).

Badokas, 2024-07-23:
@mcuxsusan any update on this issue?

SuperHeroAbner, 2024-07-24:
Hi @escherstair and @Badokas , https://github.com/nxp-mcuxpresso/CMSIS_5/tree/MCUX_2.16.000 should include the DSP 1.15.0. DSP package still uses the version name 1.10.0/1.9.0 in the source code. I'm not sure is that the reason that you think it's still old version package. Did you compared the content?

escherstair, 2024-11-05:
Hi @SuperHeroAbner,
hi @mcuxsusan 

> Did you compared the content?

Yes, I compared the content and I confirm that DSP has **NOT** been updated to 1.15.0
As an example you can check the source file `arm_rfft_fast_init_f32.c`

SuperHeroAbner, 2024-11-05:
Hi @escherstair , did you compare these files?
https://github.com/nxp-mcuxpresso/CMSIS_5/blob/MCUX_2.16.000/DSP/Source/TransformFunctions/arm_rfft_fast_init_f32.c
https://github.com/ARM-software/CMSIS-DSP/blob/v1.15.0/Source/TransformFunctions/arm_rfft_fast_init_f32.c

escherstair, 2024-11-05:
Hi @SuperHeroAbner 
> did you compare these files? https://github.com/nxp-mcuxpresso/CMSIS_5/blob/MCUX_2.16.000/DSP/Source/TransformFunctions/arm_rfft_fast_init_f32.c https://github.com/ARM-software/CMSIS-DSP/blob/v1.15.0/Source/TransformFunctions/arm_rfft_fast_init_f32.c

No. I compared the file in `ARM-software/CMSIS-DSP`
[https://github.com/ARM-software/CMSIS-DSP/blob/v1.15.0/Source/TransformFunctions/arm_rfft_fast_init_f32.c](https://github.com/ARM-software/CMSIS-DSP/blob/v1.15.0/Source/TransformFunctions/arm_rfft_fast_init_f32.c)
with the file included in the SDK 2.16.000 that I generated and downloaded from [MCUXpresso SDK Builder](https://mcuxpresso.nxp.com/en) for SDK_2.16.000_MIMX8ML8xxxKZ on 23/7/2024

And this is different from your link in github

SuperHeroAbner, 2024-11-06:
Hi @escherstair , I check the with this [package](https://mcuxpresso.nxp.com/download/df66d5b741d0782214b433ad4903cc06). It's DSP 1.15.0. Can you give me your package link?
[image]

escherstair, 2024-11-06:
Hi @SuperHeroAbner 
[image]

Here is the link of [my package](https://mcuxpresso.nxp.com/download/9d23c4dfae4479b247a199fad1ba6f97).
It's:

- ARM GCC / MCUXpresso for VS Code
- CMSIS DSP Library, FreeRTOS, multicore
- generated on 2024-07-23

SuperHeroAbner, 2024-11-06:
Hi @escherstair , I checked, no difference for source code compared with my package and the https://github.com/ARM-software/CMSIS-DSP/tree/v1.15.0

escherstair, 2024-11-06:
Sorry @SuperHeroAbner 
my fault.
There are a couple of fixes (suggested by me) added to CMSIS-DSP, but they're in the V1.16.2
These fixes are not available in the V1.15.0

I confirm that SDK 2.16.0 contains CMSIS-DSP V1.15.0.

I hope from now onwards every new NXP SDK will contain the up-to-date CMSIS-DSP (and so V1.16.2 at the moment of writing).

closed 2024-11-06 by escherstair