nxp-mcuxpresso/legacy-mcux-sdk #162: caddr_t is not available in modern toolchains

karlp, 2024-01-19:
**Describe the bug**
The type name 'caddr_t' is used in [fsl_sbrk.c.](https://github.com/nxp-mcuxpresso/mcux-sdk/blob/main/utilities/misc_utilities/fsl_sbrk.c#L24)

ARM gcc 12.x will not recognize this as a valid type.  it's wildly deprecated, you should stop using this.

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: 2.11.0
  - Toolchain: gcc version 12.3.1 20230626 (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)) 
  - Board/SoC: FRDM-k66f
  - Example in this case, /usb_examples/usb_device_phdc_weighscale/freertos but it doesn't matter
- Steps to reproduce the behavior:
  1. cd frdmk66f/usb_examples/usb_device_phdc_weighscale/freertos but it doesn't matter
  2. n7a [optional] If modified code is needed, please provide the modified code snippet. 
  3. Run build_debug.bat to build example, see error. Skip the below steps for run error.

**Expected behavior**
compiles

**Screenshots and console output**
```
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c:22:1: error: unknown type name 'caddr_t'
   22 | caddr_t _sbrk(int incr);
      | ^~~~~~~
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c:23:1: error: unknown type name 'caddr_t'
   23 | caddr_t _sbrk(int incr)
      | ^~~~~~~
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c: In function '_sbrk':
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c:29:5: error: unknown type name 'caddr_t'
   29 |     caddr_t ret;
      |     ^~~~~~~
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c:42:16: error: 'caddr_t' undeclared (first use in this function)
   42 |         ret = (caddr_t)-1;
      |                ^~~~~~~
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c:42:16: note: each undeclared identifier is reported only once for each function it appears in
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c:48:24: error: expected ';' before 'prev_heap_end'
   48 |         ret = (caddr_t)prev_heap_end;
      |                        ^~~~~~~~~~~~~
      |                        ;
/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c:28:11: warning: variable 'prev_heap_end' set but not used [-Wunused-but-set-variable]
   28 |     char *prev_heap_end;
      |           ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/dev_phdc_weighscale_freertos.elf.dir/build.make:663: CMakeFiles/dev_phdc_weighscale_freertos.elf.dir/home/karlp/src/SDK_2_11_0_FRDM-K66F/devices/MK66F18/utilities/fsl_sbrk.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/dev_phdc_weighscale_freertos.elf.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
karlp@leram-marel-net:~/src/SDK_2_11_0_FRDM-K66F/boards/frdmk66f/usb_examples/usb_device_phdc_weighscale/freertos/armgcc (main)$
```

mcuxsusan, 2024-01-24:
Hi @karlp, thanks for reporting the issue. I have already reported the issue to development team, feedback maybe delayed.

zejiang0jason, 2024-01-30:
Hi @karlp , this has been resolved by: 197fbf8122ca960ad7517056ec938ea5871c13e5. please try, thanks.

karlp, 2024-01-30:
Personally, I think that's a horrible solution.  caddr_t is not a posix sys/types.h type, (see https://pubs.opengroup.org/onlinepubs/000095399/basedefs/sys/types.h.html)  you've just continued using this wildly old caddr_t instead of fixing the source.  It depends on __USE_MISC, from _BSD_SOURCE or _SVID_SOURCE.

Why don't you just fix this single use of this old style type?

Still, thanks for at least looking at it I guess...

zejiang0jason, 2024-01-30:
Thanks @karlp , what is your suggest fixing?

zejiang0jason, 2024-01-30:
Hi @karlp , I checked in the new gcc libc, the prototype is already changed to `void *_sbrk (ptrdiff_t incr)`, so I will change accordingly, and also, seems the _sbrk is already updated for the heap_limit comparasion issue, if so, then the _sbrk in sdk is not needed. I will evaluate and update.
Thanks.

zejiang0jason, 2024-02-07:
Hi @karlp , the patch is ready in: e345b8b. thanks.

closed 2024-02-17 by zejiang0jason