nxp-mcuxpresso/legacy-mcux-sdk #186: [BUG] DMA XFERCFG::XFERCOUNT overflow when submitting large I2S transfer

maciejmatuszak, 2024-04-04:
**Describe the bug**
Submitting large I2S DMA Rx transfer using I2S_RxTransferReceiveDMA(...) splits the transfer into multiple DMA transfers. This happens if I2S_GetTransferBytes() reduce the size. I2S_GetTransferBytes calculates max single DMA transfer which is limited by hardware registers XFERCFGn:XFERCOUNT and XFERCFGn:WIDTH however the function I2S_GetTransferBytes uses:
```
#define DMA_MAX_TRANSFER_BYTES (DMA_MAX_TRANSFER_COUNT * sizeof(uint32_t))
``` 
Which assumes frame width of 4 bytes `(sizeof(uint32_t))`.
This leads to I2S_GetTransferBytes() limiting it to 4096 bytes. The HW register XFERCFGn is set by DMA_SetChannelXferConfig()
which tries to set the  XFERCFGn:XFERCOUNT field to 4096/2=2048 which is bigger than the 10 bits the XFERCFGn:XFERCOUNT allows. As a result only half of the 4096 bytes buffer to be filled. 

**To Reproduce**

- Environment (please complete the following information):
  - Tag/Commit hash: MCUX_2.14.0
  - Toolchain: arm-none-eabi-g++ (15:10.3-2021.07-4) 10.3.1 20210621 (release)
  - Board/SoC: LPCXpresso55S69
  - Example: lpcxpresso55s69_i2s_dma_record_playback
- Steps to reproduce the behavior:
 1. [set the s_Buffer to 4096 bytes](https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/2214970efbedd913e47ea0fa2ee530d8a9ba7a7a/lpcxpresso55s69/driver_examples/i2s/dma_record_playback/cm33_core0/i2s_dma_record_playback.c#L77):  
```
__ALIGN_BEGIN static uint8_t s_Buffer[4096] __ALIGN_END;
```
 2. [Update the s_RxConfig.oneChannel = true; This will set the frame length yo 2 bytes](https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/2214970efbedd913e47ea0fa2ee530d8a9ba7a7a/lpcxpresso55s69/driver_examples/i2s/dma_record_playback/cm33_core0/i2s_dma_record_playback.c#L241)
```
    I2S_RxGetDefaultConfig(&s_RxConfig);
    s_RxConfig.divider     = DEMO_I2S_CLOCK_DIVIDER;
    s_RxConfig.oneChannel = true;
    s_RxConfig.masterSlave = DEMO_I2S_RX_MODE;
```
 3. [after the `s_RxTransfer` is updated set the `s_Buffer` to known values using memset](https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/2214970efbedd913e47ea0fa2ee530d8a9ba7a7a/lpcxpresso55s69/driver_examples/i2s/dma_record_playback/cm33_core0/i2s_dma_record_playback.c#L271)
```
    s_RxTransfer.data     = &s_Buffer[0];
    s_RxTransfer.dataSize = sizeof(s_Buffer);
    memset(s_Buffer, 'r', sizeof(s_Buffer));
```
 4. Put breakpoint in [RxCallback](https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/2214970efbedd913e47ea0fa2ee530d8a9ba7a7a/lpcxpresso55s69/driver_examples/i2s/dma_record_playback/cm33_core0/i2s_dma_record_playback.c#L297)
 5. Execute the code till the RxCallback is called
 6. Examine the `s_Buffer` only half of it has data second half has the original `r` characters

**Expected behavior**
The `s_Buffer` should be filled completly

**Screenshots and console output**
[image]

**Additional context**

maciejmatuszak, 2024-04-04:
This is How I fixed it:
```
static uint16_t I2S_GetTransferBytes(i2s_dma_handle_t *handle, volatile i2s_transfer_t *transfer)
{
    assert(transfer != NULL);

    uint16_t transferBytes;

    if (transfer->dataSize >( DMA_MAX_TRANSFER_COUNT*handle->bytesPerFrame))
    {
        transferBytes = DMA_MAX_TRANSFER_COUNT*handle->bytesPerFrame;
        if ((transferBytes % 4U) != 0U)
        {
            transferBytes -= (transferBytes % 4U);
        }
    }
    else
    {
        transferBytes = (uint16_t)transfer->dataSize;
    }

    return transferBytes;
}
```

Also I put assert in `DMA_SetChannelXferConfig`
```
static inline uint32_t DMA_SetChannelXferConfig(
    bool reload, bool clrTrig, bool intA, bool intB, uint8_t width, uint8_t srcInc, uint8_t dstInc, uint32_t bytes)
{
    assert(((uint32_t)bytes / (uint32_t)width - 1UL) < DMA_MAX_TRANSFER_COUNT);
    return (DMA_CHANNEL_XFERCFG_CFGVALID_MASK | DMA_CHANNEL_XFERCFG_RELOAD(reload) |
            DMA_CHANNEL_XFERCFG_CLRTRIG(clrTrig) | DMA_CHANNEL_XFERCFG_SETINTA(intA) |
            DMA_CHANNEL_XFERCFG_SETINTB(intB) |
            DMA_CHANNEL_XFERCFG_WIDTH((uint32_t)width == 4UL ? 2UL : ((uint32_t)width - 1UL)) |
            DMA_CHANNEL_XFERCFG_SRCINC((uint32_t)srcInc == 4UL ? ((uint32_t)srcInc - 1UL) : (uint32_t)srcInc) |
            DMA_CHANNEL_XFERCFG_DSTINC((uint32_t)dstInc == 4UL ? ((uint32_t)dstInc - 1UL) : (uint32_t)dstInc) |
            DMA_CHANNEL_XFERCFG_XFERCOUNT((uint32_t)bytes / (uint32_t)width - 1UL));
}
```

mcuxsusan, 2024-04-15:
Thanks for raising up the issue and suggest fix, I have forwarded the issue to internal developer, but the feedback maybe delayed, appreciate for your patience.