nxp-mcuxpresso/legacy-mcux-sdk #192: [BUG] evkbmimxrt1170_dev_composite_hid_audio_unified_freertos_cm7 works in 2.14, does not work in 2.15

orlyprofili, 2024-05-02:
**Describe the bug**
with 2.14, this example just works
with 2.15, the device appears on usb as an audio device, and apps can play to it, but no sound

**To Reproduce**
try the example in 2.14 and 2.15 and listen to the headphone output.

- Environment (please complete the following information):
vscode on mac m silicon, rt1170 evkb

**Additional context**
I did a folder compare and even pushed every difference in the .h and .c files from 2.14 onto 2.15.  even all the other files (build related) were ensured to be the same aside from path differences as files were reorganized from sdk 2.14 to 2.15
Because of this, I believe the issue lies in the sdk, not in the example, not clear to me yet where.

mcuxsusan, 2024-05-06:
Thanks for reporting this, already forwarded to development team for check. Appreciate your patience since reply could be delayed.

AidenHu, 2024-08-22:
Hi @orlyprofili,
Something need to confirm with you. 
1. Do you have any code change for dev_composite_hid_audio_unified_freertos demo in SDK? 
2. Which host do you use?  Do you try to test this case under different USB host?
3. Which toolchain do you use? MCUXpresso or IAR?

orlyprofili, 2024-08-23:
@AidenHu 
- take example as is
- same result in macos and linux
- vscode+mcux extension+cmake
- to test this problem: 

    - west update to 2.14 and get the mentioned example. you will hear audio in the headphone jack of the rt1170 evkb
    - west update to 2.15, 2.16 or main and you will not hear audio
    - in all cases, the device will enumerate
    - it doesn't matter if you delete the example and get fresh when you switch sdk, except for a spelling error issue in 2.14. the result is the same
    
Aside: it seems debugging the multicore RT MCU with vscode is quite awkward with vscode, is there any documentation or guidance for debugging across cores?

orlyprofili, 2025-03-17:
Any update on this? I have tried with the current release of Preview 2 for MCUXpresso SDK 25.03.00 release of the newly reorganized mcuxsdk and the problem persists.  Seems mcu-sdk v2.14 is the only version where you can hear audio and I reconfirmed this. Anything higher than 2.14 or the new mcuxsdk, and it will enumerate and play a stream, but nothing audible from the headphone jack.

note also with rt1170evkb and Preview 2 for MCUXpresso SDK 25.03.00, armgcc 13.2, vscode extension on macos:
- the dev_audio_speaker_freertos_cm7 does seem to work
- maestro speaker does seem to work

unfortunately these all have quite significant differences in the way the usb audio device is implemented, so it's not obvious to me which i should reference as the "gold standard" from which to build on?

ps - would be really cool if tinyusb could be a component in the mcuxsdk ecosystem, it's really well formed and easy to work with.

AidenHu, 2025-03-19:
@orlyprofili 

Sorry that I do not response it in time.  I will check it recently. More question, do you try to test it on Windows OS?

AidenHu, 2025-03-31:
Hi @orlyprofili,

I have checked the 25.03 SDK and also confirmed with our test team, there is no sound issue for hid audio unified case on evkb1170 using MCUX SDK code on bitbucket.
From my understanding, the code on github should be totally same with the one on bitbucket (MCUX SDK).

orlyprofili, 2025-04-06:
@AidenHu I can confirm that both audio speaker and audio+hid composite work fine on Windows.

orlyprofili, 2025-04-06:
@AidenHu I'll also confirm that when you stop the debugger, the usb audio continues but with a crackling sound.  This in itself is not a major concern, but does speak to possible normal use conditions where the usb device could get into such a state, and not gracefully make its way out of it.